SatDump/SatDump #719: Running After the Android project, APP crashes directly

tianguo1314, 2024-05-23:
error message：

Fatal signal 11 (SIGSEGV), code 1 (SEGV_MAPERR), fault addr 0x0 in tid 28814 (Thread-6), pid 28757 (satdump.SatDump)
Cmdline: org.satdump.SatDump
pid: 28757, tid: 28814, name: Thread-6  >>> org.satdump.SatDump <<<
      #01 pc 00407b6d  /data/app/~~4h4UCSh0gXFd9IrPt5LaTw==/org.satdump.SatDump-h688GfVjQjax_x72_JGr8w==/lib/arm/libsatdump_core.so (loadPlugin(std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> >)+232) (BuildId: e7bba7d6842ec866937c2d1673f741b9fc4c5bc1)
      #02 pc 004081ab  /data/app/~~4h4UCSh0gXFd9IrPt5LaTw==/org.satdump.SatDump-h688GfVjQjax_x72_JGr8w==/lib/arm/libsatdump_core.so (loadPlugins(std::__ndk1::map<std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> >, std::__ndk1::shared_ptr<satdump::Plugin>, std::__ndk1::less<std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > >, std::__ndk1::allocator<std::__ndk1::pair<std::__ndk1::basic_string<char, std::__ndk1::char_traits<char>, std::__ndk1::allocator<char> > const, std::__ndk1::shared_ptr<satdump::Plugin> > > >&)+958) (BuildId: e7bba7d6842ec866937c2d1673f741b9fc4c5bc1)
      #03 pc 005f1c7b  /data/app/~~4h4UCSh0gXFd9IrPt5LaTw==/org.satdump.SatDump-h688GfVjQjax_x72_JGr8w==/lib/arm/libsatdump_core.so (satdump::initSatdump()+1530) (BuildId: e7bba7d6842ec866937c2d1673f741b9fc4c5bc1)
      #04 pc 0000df95  /data/app/~~4h4UCSh0gXFd9IrPt5LaTw==/org.satdump.SatDump-h688GfVjQjax_x72_JGr8w==/lib/arm/libandroid_imgui.so (init(android_app*)+720) (BuildId: 05c9d0d3a0f41d74ce080dcff68acc5e573548e2)
      #05 pc 00015ee1  /data/app/~~4h4UCSh0gXFd9IrPt5LaTw==/org.satdump.SatDump-h688GfVjQjax_x72_JGr8w==/lib/arm/libandroid_imgui.so (BuildId: 05c9d0d3a0f41d74ce080dcff68acc5e573548e2)
      #06 pc 0000ea03  /data/app/~~4h4UCSh0gXFd9IrPt5LaTw==/org.satdump.SatDump-h688GfVjQjax_x72_JGr8w==/lib/arm/libandroid_imgui.so (android_main+158) (BuildId: 05c9d0d3a0f41d74ce080dcff68acc5e573548e2)
      #07 pc 00015e07  /data/app/~~4h4UCSh0gXFd9IrPt5LaTw==/org.satdump.SatDump-h688GfVjQjax_x72_JGr8w==/lib/arm/libandroid_imgui.so (BuildId: 05c9d0d3a0f41d74ce080dcff68acc5e573548e2)
------------------------------------------------------------------------------------------------------------------------------------------

JVital2013, 2024-05-25:
This should be fixed as of the latest commit. If not, let us know!

tianguo1314, 2024-05-29:
The latest version I have downloaded currently does not seem to have any fixed issues. After commenting on the libsatdump_interface.so plugin, the application can run normally without any errors. Do I want to know what the potential risks are? If so, what is it?

tianguo1314, 2024-05-29:
At present, the latest version I have downloaded does not seem to have fixed this issue. After commenting on the libsatdump interface. so plugin, the application can run normally without any errors. Do I want to know what the hidden dangers are? If so, what is it?

JVital2013, 2024-05-29:
Sorry, I don't know what you mean by "After commenting on the libsatdump interface".

Either way, please verify that you have tried this version of Satdump-debug.apk, as it is currently the latest - https://github.com/SatDump/SatDump/actions/runs/9234937396. If this crashes for you, I will need the entire logcat

tianguo1314, 2024-05-30:
I blocked one of the plugins that I added and it can run normally, but I don't know what the impact is? The APK I downloaded on GitHub can run normally, so I'm not sure if I made some mistakes while packaging the static library. Can you provide a correct static library?

--------------------------------------------------------------------------------------------------------------------------------------------
If it is "libsatdump interface. so", it will not load this static library anymore  

plugin.cpp:
  if (path.find("libsatdump_interface") == std::string::npos) {
                    LOGD("The android path is: %s", path.c_str());
                    std::shared_ptr<satdump::Plugin> pl = loadPlugin(path);
                    loaded_plugins.insert({pl->getID(), pl});
  }

tianguo1314, 2024-05-30:
[image]

tianguo1314, 2024-05-30:
The runtime error caused by this being empty

JVital2013, 2024-05-30:
Correct, it should not be a plugin.

You are using an old codebase - this has already been fixed in commit 8b877bb - you can see I implemented this exact fix there.

commit c1c6587d (2024-05-25): Fix #719
closed 2024-05-30 by JVital2013