SatDump/SatDump #720: Tracking scheduler load/save configuration

kng, 2024-05-24:
Using the tracking scheduler and reconfiguring it often can be a bit tedious.
Adding a load/save feature to be able to manage the recorder_tracking section directly in the ui.
It will make it easier to provide templates/examples in the repo as well.

It is possible to cut/paste in that section in the config, but having it in the ui would be nice.
It would be nice if the format was compatible with the [cli](https://docs.satdump.org/autotrack.html#cli-configuration) as well, but there seems to be some differences in naming.

Aang23, 2024-06-09:
Some sort of "export CLI configuration" could be added perhaps, it'd be simple enough as most configuration sections are by design identical. Only some more implementation specific things vary between both.

JVital2013, 2024-10-06:
This has been implemented in the latest nightly, and in 1.2.1 (due to be released soon).

[image]

closed 2024-10-06 by JVital2013