SatDump/SatDump #721: Overriding the default configuration

matveyhim, 2024-05-24:
**Description of the issue**
I wanted to disable the MCIR autogen and add some projections, but changing satdump_cfg and settings.json does not work.
**Hardware (SDR/PC/OS)**
Ubuntu 22.04
**Version (Eg, 1.0.0, CI Build 171)**
SatDump v1.2.1-59380c0c
**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
viewer array in settings.json file
```
"viewer": {
        "instruments": {
            "avhrr_3": {
                "MCIR (Uncalibrated)": {
                    "autogen": false
                },
                "APT channel A": {
                    "draw_map_overlay": true,
                    "project": {
                        "width": 4096,
                        "height": 2048,
                        "draw_map_overlay": true,
                        "config": {
                            "type": "stereo",
                            "lat":60,
                            "lon": 77,
                            "scale": 1000
                        }
                    }
                }
            }
        },
```

matveyhim, 2024-05-25:
No, it doesn't look like it. If I miss any `}` character in the files, satdump will crash. Or if I add the msu_mr section in settings.json, it will be sorted to the end of the instruments section

matveyhim, 2024-05-27:
Strangely, the projection does not work with bad images, but it works with good ones that I received today.

Aang23, 2024-06-09:
Are you receiving APT or HRPT? If using APT all images good or bad should project. If it's HRPT very low signal levels can how problems and there's nothing to do about it.

matveyhim, 2024-06-09:
It APT. And I had another bug on windows - the map overlay that I do in the viewer was strange, but the projections are normal

matveyhim, 2024-06-09:
It's APT. I also have a bug in Windows - the map overlay that I'm doing in the Viewer is getting weird, but the projections remain normal. Everything is fine in ubuntu too. There's nothing wrong with the names of the satellites I selected.

JVital2013, 2024-07-29:
If you are still having an issue about projection or map overlays, please open another issue to keep things tidy. Having multiple issues in a single ticket causes confusion.

As for your original issue (overriding the default configuration), please try again with the latest build. "Advanced mode" has been implemented, which lets you do this from the GUI. Behind-the-scenes, advanced mode works the same way you were trying to override the settings.

If you are still having issues with overriding the config, please post:
- Your entire satdump_cfg.json, and the path you copied it from
- Your entire settings.json, and the path you copied it from
- The entire log file from an instance where your custom settings were not applying

JVital2013, 2024-09-29:
@matveyhim are you still having issues overriding the default configuration?

matveyhim, 2024-09-30:
I tested it today, turned off the projection of ch4 from the advanced settings, everything worked properly

JVital2013, 2024-09-30:
Great, glad to hear it!

closed 2024-09-30 by JVital2013