SatDump/SatDump #723: Improve program documentation available offline

pkreuzt, 2024-05-30:
Satdump's documentation is severely lacking in content for the programs themselves. The only relevant info is on the wiki, but you don't always have internet connectivity available when out on the field.

It would be good to have options (-h or --help) explaining a bit more than the basic usage and even a man page for Linux systems. For example, the available options on the command line and the possible values for those options. Man page could be more exhaustive, explaining with a certain detail what those options do.

For the UI it would be possible to package the whole wiki on a separate doc package to browse it offline, like in Linux distros. Links on About page to browse the wiki could then open the local resources in a browser like most GUI programs do.

Aang23, 2024-06-09:
To be honest, the first thing to do would be to improve documentation period. But even though we keep saying this it's not that easy to do.

Bundling the generated HTML could be done, but would balloon downloads for everyone so perhaps a separate downloadable ZIP on the Documentation repo could be a better option.

pkreuzt, 2024-07-01:
I was thinking more in the way of a cheat sheet. Most of the time we just need something to remember what was the name of the option or the format/range of the parameters to use. This is what you'd expect from a basic -h switch on the command line.

alsldlflglhljlk, 2024-08-08:
@Aang23 is there a GNU / Linux manpage for SatDump? If not, that would probably be the first step to take.