SatDump/SatDump #724: No output using WAV for NOAA

Akito01, 2024-05-31:
Using the CLI of Satdump at version 1.2.0 I can't seem to produce any output for the NOAA APT pipeline.
The input file is a 11025Hz file. 
This is part of the log output:
```
[09:36:44 - 31/05/2024] (D) Input file (--satellite_number) : 18
[09:36:44 - 31/05/2024] (D) Output file : /home/rob/ramweather/test/NOAA1820240531-110641.wav
[09:36:44 - 31/05/2024] (D) Starting noaa_apt
[09:36:44 - 31/05/2024] (I) Data is already at level baseband, skipping
[09:36:44 - 31/05/2024] (I) Data is already at level audio_wav, skipping
[09:36:44 - 31/05/2024] (I) Data is already at level products, skipping
[09:36:44 - 31/05/2024] (I) Done! Goodbye
```
This is the synax I'm using:
`satdump noaa_apt --satellite_number 18 /home/NOAA1820240531-110641.wav /home/test`

File of course exists in the folder and the test-folder exists as well. No clue what I'm doing wrong here.

JVital2013, 2024-06-01:
Try `audio_wav` instead of wav

JVital2013, 2024-06-05:
Glad to help!