SatDump/SatDump #725: Orbcomm Auto Plotter Bug and Noaa missing Image

Stepman55, 2024-06-03:
'Orbcomm STX Auto Plotter' does not show any satellites in the plotter window. In some versions before, they were still visible but disappeared again immediately and in 'Last Epeh' no time was shown but 'IIp'. Before the various variables were changed to Double, everything was fine.

Noaa Image 'Products', RGBComposites/Mcir Rain (uncal.), Only shows the color Layer  but without the satellite picture belonging.

Aang23, 2024-06-09:
Can you send over some logs? I'll try to reproduce but I would like to confirm something first.

Stepman55, 2024-06-09:
the log:
[satdump-20240607T190947.log](https://github.com/user-attachments/files/15751574/satdump-20240607T190947.log)

Stepman55, 2024-08-01:
C1.2.1-0be2236 Portable 
The lines in the plot are now visible again. Only the display for LAST EPEH constantly shows "IIu s". (look at it, not "IIp")
[image]