SatDump/SatDump #726: Noaa missing Image

Stepman55, 2024-06-05:
V 1.2.1-3e07abe - Win11
Noaa Image 'Products', RGBComposites/Mcir Rain (uncal.), Only shows the color Layer but without the satellite picture belonging.

Stepman55, 2024-06-09:
a few pictures and the corresponding log
[image]
[image]
[satdump-20240609T132737.log](https://github.com/user-attachments/files/15751652/satdump-20240609T132737.log)

Stepman55, 2024-06-30:
Recording live or loading makes no difference.
The data comes from the still functioning version 1.2.0-1e40ef1. However, the log is no longer available there.

JVital2013, 2024-07-29:
Is this still happening to you? If so, could you please share an example dataset? (Zip up the entire output folder for a single pass that SatDump generates, and upload it here)

Stepman55, 2024-08-01:
Yes
here is the result.
You can easily produce it yourself. Simply receive a short APT noise image manually. Go to the viewer, select MCIR Rain and now you will see it, ...not.
All other channels show the noise image.
[2024-08-01_06-47_noaa_apt_137.62 MHz.zip](https://github.com/user-attachments/files/16451834/2024-08-01_06-47_noaa_apt_137.62.MHz.zip)
[satdump-20240731T215242.log](https://github.com/user-attachments/files/16451851/satdump-20240731T215242.log)

alsldlflglhljlk, 2024-08-03:
I do believe that MCIR rain strips clouds from the image and places them on an internal map. Because your image isn't really a real image, it does nothing.

JVital2013, 2024-08-03:
Ah, I see what the issue is. The bug happened for you, and not me, because it only happens if the product image format is set to jpeg in settings.

I'll push a fix in a moment; you can expect a working nightly in ~30 mins!

[image]

commit 19670465 (2024-08-03): Fix #726
closed 2024-08-03 by JVital2013