SatDump/SatDump #727: MetOp NOSYNC

ArgoNavi, 2024-06-08:
I'm having an issue with MetOp not achieving sync when using autotrack despite having plenty of SNR, If I stop/restart autotrack during the pass it immediately achieves sync and starts decoding.

before restart:
[satdump-20240607T171805.log](https://github.com/user-attachments/files/15746363/satdump-20240607T171805.log)

after restart:
[satdump-20240607T231535.log](https://github.com/user-attachments/files/15746364/satdump-20240607T231535.log)

Aang23, 2024-06-08:
Can you show a screenshot?

ArgoNavi, 2024-06-08:
I’ll take one during the next pass.

ArgoNavi, 2024-06-09:
before restarting the decoder:
[image]

after restarting:
[image]

Aang23, 2024-06-09:
Are you using a RTL-SDR by any chance? If so, some problems are not unexpected with the default values as it's on the edge. If this is the case I can tell you what to change to improve things.

ArgoNavi, 2024-06-09:
Yes I'm using an RTL-SDR Blog V4, what settings should I change?

ArgoNavi, 2024-06-13:
Solved by increasing my sample rate to 2.88 and enabling both dc block and agc.

[image]

closed 2024-06-13 by ArgoNavi