SatDump/SatDump #731: V1.1 and V1.2 and latest V1.2 nightly crashes after  TLES Loaded

chanrobi, 2024-06-17:
**Description of the issue**
Tried to capture some meteor weather satellite tonight and got a "wrap GPU error", never seen this before. And I have had this working before. This was v1.1.4. 

Downloaded official v1.2.0 build, installed that now satdump only gets to "10030 TLES loaded" messsage, then this disappears and nothing happens. 

Downloaded the latest nightly V1.2.0 build, as of right now. Same issue. 

V1.1.4 was working no issues before tonight, not sure what happened. Now doesn't start at all

**Hardware (SDR/PC/OS)**
RTL SDR v4, did not have it connected when trying v1.2.0 and the nightly. Win 10 64 bit on a T480s laptop running i5 8350u.
**Version (Eg, 1.0.0, CI Build 171)**

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[satdump-20240616T224212.log](https://github.com/user-attachments/files/15862352/satdump-20240616T224212.log)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

chanrobi, 2024-06-17:
> Delete your settings.json file and try again

I deleted the settings.json file and no bueno, exact same thing still happening.

chanrobi, 2024-06-18:
I have now narrowed down the issue (after removing .dll's in the plugins folder) to the udp_source_support.dll

I followed the instructions in this post: https://github.com/SatDump/SatDump/issues/638#issuecomment-1974847378

Without this file it loads properly. Only this dll causes issues. I also updated the visual c++ redist and it made no difference

So now the question what is this file and do I need it? How to fix error ?

JVital2013, 2024-06-18:
Ah, that's the problem: udp_source_support.dll no longer exists (it has been rolled into net_source_support.dll). If you still had udp_source_support.dll, it must have been left over from a previous install. Deleting it is the right course of action and you will have no loss in functionality, since it should have been removed anyway.

closed 2024-06-18 by JVital2013