SatDump/SatDump #736: Aqua & Proba S-Band

benb0jangles, 2024-06-22:
Hello friend, please can you tell me if noaa, aqua and proba s-band are possible to live process? I am going to use mmds downconverter, i think noaa s-band using mmds is 249.5Mhz. Thanks!

Aang23, 2024-06-26:
Yes they can, however Aqua will only provide telemetry, no imagery.

closed 2024-06-26 by Aang23

benb0jangles, 2024-07-08:
Hi again, for NOAA s-band, do I record baseband then offline process?

Under recording tab:
Recorder > processing > search pipelines > NOAA HRPT
It is set for L-band 1698Mhz but I want to try 2247Mhz With same pipeline. Thank you for advice.

benb0jangles, 2024-07-14:
I just upgraded android saydimp to 1.2.0 and now I can adjust HRPT NOAA frequency to s-band 2247.5Mhz thanks

benb0jangles, 2024-07-19:
Noaa19 s-band HRPT first image yesterday:
Satdump 1.2.0
Airspy mini
Mmds
1m dish + helical
[image]

I must work on improving average dB as it is low, either mmds is weak, helical is not tuned, I will figure it out. Perhaps an LNA before downconverter? Or after? 

I was meaning to ask if a feature could be implemented for 1998 downconverter, a check ✔️ box perhaps, that will adjust frequency more easily then trying to type in as on android phone it is small screen space and typing difficult to do when setting up for Rx pass. 

Thanks 😊👍🏻