SatDump/SatDump #737: Rsp1A don't start

Vito6, 2024-06-27:
After a Satdump crash, the program don't reconizs the sdr, but the sdr is here. I need to restar the pc 

Sdrplay Rsp1A, Win 11

Satdump last ufficial

[image]

[satdump-20240625T111455.log](https://github.com/user-attachments/files/16022298/satdump-20240625T111455.log)
[satdump-20240625T085049.log](https://github.com/user-attachments/files/16022300/satdump-20240625T085049.log)

alsldlflglhljlk, 2024-06-30:
Have you tried with another SDR program like SDR++, SDRUno, or SDRconnect?

BI4PYM, 2024-07-03:
Try to restart the SDRPlay API service.

JVital2013, 2024-07-29:
This sounds like it might be a bug in the SDRPlay API. Trying SDR++, SDRUno, or SDRconnect once SatDump is in a "stuck" state to see if they work would be a good idea to see if that is the case.