SatDump/SatDump #738: Dots in demodulator constellation window are out of bounds

dvdesolve, 2024-06-28:
**Description of the issue**
I've noticed for APT and LRPT mode that some dots are out of black box boundaries (see screenshot below)
[image]

**Hardware (SDR/PC/OS)**
RTL-SDR blog v3 dongle
Arch Linux x86_64 rolling

**Version (Eg, 1.0.0, CI Build 171)**
latest git version

alsldlflglhljlk, 2024-06-30:
Sure, but is this really a bug? It doesn't impair SatDump's ability to decode and demodulate satellite signals.

dvdesolve, 2024-06-30:
of course, that's not a fatal bug. I've just chosen one out of three templates during posting the issue

JVital2013, 2024-07-01:
It's still a bug in my book - fixed as of the latest nightly! Thanks for the poke to finally fix this

closed 2024-07-01 by JVital2013