SatDump/SatDump #740: DC Carrier on empty

alsldlflglhljlk, 2024-06-30:
**Description of the issue**

Attached to this issue is a baseband recording only containing noise (file extension changed to `.webm` to allow upload). In SDR++, the file plays as expected, but SatDump adds a spurious DC carrier in the center of the FFT.

**Hardware (SDR/PC/OS)**

Intel Core i3-4000M@2.40 GHz

Windows 10 22H2

**Version (Eg, 1.0.0, CI Build 171)**

v1.2.0

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

N / A

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

Playing through SDR++:
[image]

Playing through SatDump:
[image]

JVital2013, 2024-06-30:
The spike is in the baseband; it's visible in SDR++ if you turn IQ correction off

[image]

alsldlflglhljlk, 2024-07-01:
Is there such an IQ correction option in SatDump? I've tried turning on the "DC Blocking" switch in the demodulation options, but that doesn't seem to remove the spike in the FFT.

alsldlflglhljlk, 2024-07-01:
Alright, good to know. Thanks!

closed 2024-07-01 by alsldlflglhljlk