SatDump/SatDump #741: write satdump PID when logging output

creas002, 2024-06-30:
I'm finding that RN2 logs satdump output to its run (output.log) and I find that I at times have more than one satdump process running/processing and multiple satdump processes are dumping data together at once. I'm used to this sort of things when looking at OS or Oracle logs and the solution used there is to record the PID. When satdump on *NIX starts it could grab its own PID using something mypid=$$ and echo it out as " [${mypid}] " when satdump outputs (D, T, I) information.

Thanks for consideration

Aang23, 2024-07-01:
While I do not think this is a great fit for the original intent, as SatDump spawns many different threads with different PIDs making this more or less useless to identify a SatDump instance, this could however be very useful to differentiate log messages coming from different processes, thinking about different instances of the same module sending identical messages. A bug in one of them can be hard to track back to a specific thread / situation.

creas002, 2024-07-01:
I totally agree. I typically use ptree for monitoring and handling children thread and truss when I need to try and find out what is actually happening. Recently I identify a bug in ffmpeg using truss when trying to create spectograms, all its children would hang out idle "Forever" and I had to kill them to resume. This was a Debian 12 issue only, works fine on Debian 11.    My intended purpose for new feature was simple to be able to group satdump informational output together by the parent that wrote it out.  I appreciate your comments and its really is not any big deal that its declined.  It is not a real issue so much. When I schedule several passes back to back its gets confusing with all the data coming in and previous processes still running.  No worries Alan, It's all good!