SatDump/SatDump #743: Processing Satdump output data

littlehell, 2024-07-05:
Hello,

We are using Satdump to receive X band satellite data such as Aqua, S-NPP and NOAA-21.

Now we are able to save data and process them to *.cadu with offline processing, after this I tried some software but cannot process them to CCSDS or L0.

Could advise witch software or package I can use? Thanks!

LEDFlighter, 2024-07-07:
I don't have much experience with X-Band (none), but you can use SatDump to process everything to L0-Products. Process the CADU further to L0-Products and load the results into the viewer. Then you can save individual images and composites.

littlehell, 2024-07-08:
> I don't have much experience with X-Band (none), but you can use SatDump to process everything to L0-Products. Process the CADU further to L0-Products and load the results into the viewer. Then you can save individual images and composites.

Thanks to your kind reply.
Is there any setting to generate L0 data? After my process on Satdump UI, I can only see some single channel and composites images, there's no L0 data. And checked Settings in UI but find nothing.

Aang23, 2024-07-26:
@littlehell SatDump focuses on decoding the data while re-writing most things from scratch, in order to have something both more optimized and unified for pretty much everything. Therefore, while SatDump can produce L1B or even L2 data for some missions (and more coming), this only means providing the user calibrated & geo-referenced data.

This data is stored in "custom" simpler formats (eg, PNG of the raw L0 data, metadata to run the calibration on-demand), and therefore do not match the common hdf5 stuff you'll find on most websites, as to put it simply these are very poorly documented and are often so custom & specific to their workflow it's not really viable to implement - and the time is much better spent focusing on other things. Furthermore, hdf5 is really not a standard format per say. It's an absolute mess to implement and you're locked in using the only existing implementation...

However, all data spit out by the decoders in .cadu files or equivalent is what comes down *directly* from the satellite, and with little to no processing can be fed into the "official" software packages for each missions if you wish, leaving SatDump only to act as the demodulator. You can find some information there :
- https://directreadout.sci.gsfc.nasa.gov/home.html
- https://nwp-saf.eumetsat.int/site/
- And a few more.

Do note that for many missions supported by SatDump usually thanks to heavy reverse-engineering efforts the official packages either just don't exist or cannot be acquired by an individual easily, if at all. On more than one of them, which I cite here you'll simply never get a reply from the supposed contact Email...

littlehell, 2024-07-31:
> @littlehell SatDump focuses on decoding the data while re-writing most things from scratch, in order to have something both more optimized and unified for pretty much everything. Therefore, while [...]

Hello, thanks for your reply.
What makes me very confused before is that RT-STPS&AAPP cannot process CADU data from Satdump, it should ba a standard format and should not process with error.
After that I cheked the data processing in Satdump and find it already do pn for data, and if input RT-STPS to do pn again, it will make data into messy. Now I have disabled pn in RT-SPTS and processing is ok.
Now the issue is sloved and I will close it, thanks.

closed 2024-07-31 by littlehell