SatDump/SatDump #745: Request pipeline cancel option while processing button (+ pipeline options)

LEDFlighter, 2024-07-07:
**What would you want to be added to SatDump?**
I want to request a possibility to cancel the processing at any time while it's running (if steps take too long or you don't want to do them for now). Maybe also consider an option to select which processing steps should be executed in the automatic processing.

**Describe the feature you'd like to have**
Something like a button "abort" or "cancel" somewhere, while a processing pipeline is running (screenshot follows)
[image]

**Describe alternatives you've considered**
I don't want to quit SatDump when I want to cancel the pipeline.

**Additional context**
Currently no additional context, just answer to this request or contact me for further questions.

eswnl, 2024-07-09:
I think this is related, but some kind of cancel button while processing RDAS signals. For trial recordings, its annoying having to wait until I can input another recording.
[image]

alsldlflglhljlk, 2024-08-05:
So something like this...
[image]

LEDFlighter, 2024-08-05:
> So something like this... ![IMG_3184](https://private-user-images.githubusercontent.com/119443367/355149404-0b8f6ade-000e-4c00-bdfe-ac9030da2030.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjI4ODc5MzEsIm5iZiI6MTcyMjg4NzYzMSwicGF0aCI6Ii8xMTk0NDMzNjcvMzU1MTQ5NDA0LTBiOGY2YWRlLTAwMGUtNGMwMC1iZGZlLWFjOTAzMGRhMjAzMC5qcGVnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MDVUMTk1MzUxWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9MGRiNDdlNTFkMjJlNGI4YWJjOGFhZjU2ZTgzMjY4ODAzZWZkZjFmMDBjMzI5ZTViODQyNDMwOTc0NmFhZjhkZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.gv4OqYGWI2mVM4Z-mHCOfdfIaneRLHbHskZyP9VA4Sk)

Yes, I mean something like this.