SatDump/SatDump #746: USRP Error Messages (images not found) on OSX Images refers to script that doesn't exist.

darksidelemm, 2024-07-08:
**Description of the issue**
I'm attempting to use SatDump with a USRP B210 using the MacOS build.

On load (noting I had to run the satdump-ui binary within the appbundle from a terminal to be able to see the log!) I get log messages about the UHD images not being available. Fair enough, I wouldn't expect you to ship them with SatDump.

However I do get this message:
```
[WARNING] [B200] EnvironmentError: IOError: Could not find path for image: usrp_b200_fw.hex
Using images directory: <no images directory located>
Set the environment variable 'UHD_IMAGES_DIR' appropriately or follow the below instructions to download the images package.
Please run:
 "/Applications/SatDump.app/Contents/libs/uhd/utils/uhd_images_downloader.py"
```

The file `/Applications/SatDump.app/Contents/libs/uhd/utils/uhd_images_downloader.py` doesn't exist. In fact, the `/Applications/SatDump.app/Contents/libs/uhd/` doesn't exist either!

So, perhaps this message shouldn't be shown.

I was able to download the images using my homebrew-installed uhd scripts and then run satdump-ui using: `UHD_IMAGES_DIR=/opt/homebrew/Cellar/uhd/4.6.0.0_1/share/uhd/images ./satdump-ui`, so it's good to know that USRP support does work with the images in place.

**Hardware (SDR/PC/OS)**
Macbook Pro M1, OSX 12.7

**Version (Eg, 1.0.0, CI Build 171)**
v1.2.0, OSX Silicon Appbundle - https://github.com/SatDump/SatDump/releases/download/1.2.0/SatDump-macOS-Silicon.dmg

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
No Crash. Just pointing out the log messages above :-)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

JVital2013, 2024-07-28:
Hello @darksidelemm! First off: thanks for testing USRP SDRs on macOS! I don't have a USRP SDR, and the other devs are less active on the macOS side of things, so testing this SDR on macOS is tricky at the moment.

These error messages are coming directly from the embedded UHD library, which is why you cannot see them in the UI, unless you run it from the terminal. I just made a change to hide the message about the Python script.

Additionally, I added an option to the settings screen to control where the UHD driver looks for firmware images. Could you please verify that this works correctly for you?

[image]

darksidelemm, 2024-07-28:
Unfortunately I had to return that USRP back to where I borrowed it from! I'll see if I can get it back to do this test.

JVital2013, 2024-09-29:
I haven't heard of this feature not working, so I'll close this issue for now. If you have any issues setting the USRP firmware path, let us know though and we can take a look!

closed 2024-09-29 by JVital2013