SatDump/SatDump #747: PlutoSDR not working from CLI

fede-tripod, 2024-07-10:
Hi I am trying to use a satdump with PlutoSDR connected via USB. Using the command:

_satdump live metop_ahrpt metop_output_directory --source plutosdr --samplerate 4e6 --frequency 1701.3e6 --general_gain 18 --timeout 780 --dc_block_

I get the following error:

[image]

Trying to give the command:

_iio_readdev -u usb:usb:3.5.5 -b 100000 cf-ad9361-lpc | pv > /dev/null_

everything works correctly.

The same problem also occurs from GUI by selecting the PlutoSDR as a USB device. Instead, using it as an IP device everything works smoothly from GUI.
[image]

Is there any way to force the use of Plutosdr IP also from CLI? 
Thank you.

Translated with DeepL.com (free version)

fede-tripod, 2024-08-21:
Solved by editing the code in:
/plugins/sdr_sources/plutosdr_sdr_support/plutosdr_sdr.cpp

Line 180, from:
ctx = iio_create_context_from_uri(std::string(“usb:” + d_sdr_id).c_str());
to:
ctx = iio_create_context_from_uri(std::string(d_sdr_id).c_str());

Everything seems to be working now!

is it possible to implement this fix to the online code as well?

bujidaa, 2024-11-10:
have you ever try this on windows? my satdump cant detect plutosdr from usb