SatDump/SatDump #753: Problem using satdump on a NOAA-V2 rpi

Betacarotin, 2024-07-25:
I just build the raspberry-noaa-v2 project on a new rpi (instead of using the image they provide). Hence following their exact instructions, itś not working as it should. digging through the logs I found something strange in stadumps logs. How can I solve this? What may the problem be? 

[10:19:00 - 25/07/2024] [32m(I)    _____       __  ____                      [m
[10:19:00 - 25/07/2024] [32m(I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ [m
[10:19:00 - 25/07/2024] [32m(I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \[m
[10:19:00 - 25/07/2024] [32m(I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /[m
[10:19:00 - 25/07/2024] [32m(I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ [m
[10:19:00 - 25/07/2024] [32m(I)                                     /_/      [m
[10:19:00 - 25/07/2024] [32m(I) Starting SatDump v1.2.1-1021e84b[m
[10:19:00 - 25/07/2024] [32m(I) [m
[10:19:00 - 25/07/2024] [32m(I) Loading config /usr/share/satdump/satdump_cfg.json[m
[10:19:00 - 25/07/2024] [32m(I) Loading user settings /home/demesos/.config/satdump/settings.json[m
[10:19:00 - 25/07/2024] [32m(I) Loading plugins from /usr/lib/satdump/plugins[m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libsdrpp_server_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin spyserver_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libcpp_compos_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin standard_cpp_compos loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libjpss_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin jpss_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libfengyun4_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin fenyun4_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libdvb_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin dvb_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/librtlsdr_sdr_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin rtlsdr_sdr_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libanalog_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin analog_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/liblimesdr_sdr_support.so...[m
[10:19:00 - 25/07/2024] [31m[1m(E) Error loading /usr/lib/satdump/plugins/liblimesdr_sdr_support.so! Error : libLimeSuite.so.22.09-1: cannot open shared object file: No such file or directory => src-core/core/plugin.cpp:19[m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libproba_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin proba_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/librtltcp_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin rtltcp_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libbladerf_sdr_support.so...[m
[10:19:00 - 25/07/2024] [31m[1m(E) Error loading /usr/lib/satdump/plugins/libbladerf_sdr_support.so! Error : libbladeRF.so.2: cannot open shared object file: No such file or directory => src-core/core/plugin.cpp:19[m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libmats_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin mats_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libeos_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin eos_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libjason3_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin jason3_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libstereo_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin stereo_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libsimd_neon.so...[m
[10:19:00 - 25/07/2024] [31m[1m(E) CPU Does not support NEON. Extension plugin NOT loading![m
[10:19:00 - 25/07/2024] [37m(T) Plugin simd_neon loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libumka_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin umka_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libfengyun3_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin fengyun3_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libnoaa_metop_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin noaa_metop_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libbluewalker3_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin bluewalker3_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libairspy_sdr_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin airspy_sdr_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libmeteor_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin meteor_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/liborbcomm_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin orbcomm_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libplutosdr_sdr_support.so...[m
[10:19:00 - 25/07/2024] [31m[1m(E) Error loading /usr/lib/satdump/plugins/libplutosdr_sdr_support.so! Error : libiio.so.0: cannot open shared object file: No such file or directory => src-core/core/plugin.cpp:19[m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/libmirisdr_sdr_support.so...[m
[10:19:00 - 25/07/2024] [37m(T) Plugin mirisdr_sdr_support loaded![m
[10:19:00 - 25/07/2024] [37m(T) Loading plugin /usr/lib/satdump/plugins/li

JVital2013, 2024-07-25:
Hello!

Is this the entire log? It looks like plugin loading was stopped partway through. I'm also seeing some errors about loading plugins for SDRs like the BladeRF and PlutoSDR. This shouldn't be a problem if you aren't using those SDRs.

Some ideas:
- Are you using the same version of debian as their image? If not, that is likely the problem.
- You could try running SatDump manually and see if you get any more info.

Raspberry-noaa-v2 uses a pre-built version of SatDump from a few weeks ago (that they update regularly), and they have verified it as working on their system. It looks like you're still using their .deb, so I'm inclined to say it might better to open an issue with them if the above feedback does not help.

Betacarotin, 2024-07-25:
Hello, 
thanks for getting back to this. 
No its not the entire log, only the part with errors. the rest of the log looks good. 
I do not use BladeRF or Pluto SDR. 

- I am using Bookworm 32bit, while their image originates from Bullseye or even older if I understood the docs correct. Ahhh so this could cause the trouble?
- I can run it manually with the UI, it works fine, I can access the SDR and see signals comming in. 

So it looks like it would be best to start with their image and build the entire system based on that, than starting with a fresha and empty system and try to manually install everything else? 

(The reason behind it, is that I would like to have one rpi that runs ADS-B decoding AND noaa weather sats, with two RTL_SDR and two seperate antennas. Otherwise I will bi drowning in rpi on my desk soon. I mean it must be possible to use a pi for more than one project, especially an rpi 5 8gb)

JVital2013, 2024-07-25:
For SatDump itself, the most important thing is to match Debian versions. It looks like they do provide debs for bookworm - as long as you installed that one, it would be OK. If SatDump itself runs directly, that more makes me thing there's some compatibility/configuration issue with their scripts on your system.

There should be nothing stopping you from running raspberry-noaa-v2 and another project on the same pi. Starting with their image is a good idea, then see if you can add your other services on top.

JVital2013, 2024-09-29:
@Betacarotin would you like any further assistance from us with this issue?

Betacarotin, 2024-10-01:
Sorry for not responding! No thanks, with the right debian version it works, like you said.

closed 2024-10-01 by Betacarotin