SatDump/SatDump #754: Map overlay broken

sattrackcam, 2024-07-25:
The map overlay is broken since I updated the program.

ASUS laptop, Window 11, 
Satdump 1.2.1-23a7665

[image]
[meteor_m2-x_lrpt.cadu.txt](https://github.com/user-attachments/files/16381668/meteor_m2-x_lrpt.cadu.txt)

N1TX, 2024-07-30:
Samples from NOAA 15 (29 July 1948z) and METEOR M2-3 (29 July 2219z). Receiving location is 64.854202N -147.125000.

[image]
[image]
![avhrr_3_ch2_lat-lon_overlay](h
[image]
ttps://github.com/user-attachments/assets/b5942ef2-43a6-4d7b-b37a-9ff429fa509d)
[image]
[image]
[image]

JVital2013, 2024-07-30:
Hi @N1TX, thanks for being willing to share samples!

Instead of the processed images, could you please share either an audio .wav (for apt), .cadu (for meteor), or zipped-up satdump output folder? Unfortunately these processed images do not help us with troubleshooting.

N1TX, 2024-07-31:
[N1TX.zip](https://github.com/user-attachments/files/16444735/N1TX.zip)
METEOR CADU files are in this file. The WAV files for APT are too large to upload. What parameters do you prefer for the WAV recordings?

Cumolonimbus, 2024-12-21:
I have noticed exactly the same problem. However sometimes the overlay mapping behaves as it should, making it look like it is more or less a random problem. The "strange" lines (like country borders) appear always in the arctic ocean region around Greenland, Iceland and Scandinavia. I live in the Netherlands. 90 % of the time the NOAA (15, 18, and 19) passes are affected with this. The Meteor satellites are hardly affected. It happens with all my SDR receivers (Airspy and SDRplay). Interestingly it never happens when I use WXtoIMG. I have checked my coordinates, update with the latest Keplers, checking and setting my pc clock to the precise time and selecting and match each individual NOAA satellite number in the tracking configuration settings ad infinitum, to no avail.

Eric

[image]

Cumolonimbus, 2024-12-21:
I have noticed exactly the same problem. However sometimes the overlay mapping behaves as it should, making it look like it is more or less a random problem. The "strange" lines (like country borders) appear always in the arctic ocean region around Greenland, Iceland and Scandinavia. I live in the Netherlands. 90 % of the time the NOAA (15, 18, and 19) passes are troubled with this. The Meteor satellites are hardly affected. It happens with all my SDR receivers (Airspy and SDRplay). Interestingly it never occurs when I use WXtoIMG. I have checked my coordinates, update with the latest Keplers, checking and setting my pc clock to the precise time and selecting and matching each individual NOAA satellite number in the tracking configuration settings ad infinitum, to no avail. I use Satdump v1.2.2.

Eric

[image]

jgellison, 2025-01-10:
I have exactly this issue as well.  I am based in the UK and the map overlay for NOAA shows extra coastlines around Greenland.  It looks like a second incorrect projection has been applied and there appears to be a second set of lat/lon lines.

I'm on version 1.2.2

Baerchen239, 2025-01-17:
I can also confirm the problem...
It was like that in version 1.2.1, and now in version 1.2.2
However, the problem only exists in the map/city overlays in the products.
The maps behave correctly in the projections.