SatDump/SatDump #764: Image Autorotate

ve3elb, 2024-07-31:
**Description of the issue**

Running SatDump in full auto mode with live decode and processing. I noticed that the day images are auto rotated correctly however the late afternoon and evening images don't. 

I have done this on both windows, Linux Ubuntu and PiOS with the same results. 

On the other hand would be to much work to add some of the offline/viewer processing settings to the settings page so the user can controll which settings to choose for live decode and processing. 

**Hardware (SDR/PC/OS)**

**Version (Eg, 1.0.0, CI Build 171)**

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

alsldlflglhljlk, 2024-08-01:
What satellite / mode is this? NOAA APT / HRPT, Meteor-M HRPT, MetOp AHRPT, or something else?

ve3elb, 2024-08-01:
NOAA APT, METEOR M2 LRPT

closed 2024-08-01 by JVital2013

Aang23, 2024-08-01:
My stance on this is simple : we simply should never rotate the raw data IMHO. We'd be adding a massive level of complexity in the raw products for something that is meant to be a simpler, generic format for storing the extracted data. 

However, having an option to handle it after processing, say as a checkbox in the viewer and option on composites would be easy enough.

ve3elb, 2024-08-08:
@Aang23  this is want I ment. Auto rotate the finished processed output image. There are no settings options for auto tracking all available options are for manual processing.  Am looking to have a fully automated unattended system. Am liking how SatDump works.  

Thank you all for a great software.

t0nitogp, 2024-08-24:
Yes to at least have the option to enable auto rotate would make complete sense, it would be then up to the user if they want the image to rotate on its own or not.

Nikotine1, 2025-10-16:
Sorry for bringing to live this dead topic, but this would help me as well.
I'm running Satdump CLI in a docker container on a Raspberry Pi on the roof, fully auto-tracking satellites.
I would like to make timelapses of the images it gets, but because the images aren't rotated this doesn't make sense.