SatDump/SatDump #766: Baseband File Sizes

ve3elb, 2024-08-01:
@JVital2013 I have setup SatDump on WindowsOS, UbuntuOS and PiOS all run great. The issue is when using SatDump in full automatic mode tracking, Decoding and live processing METEOR-M2 LRPT,NOAA APT the baseband files full up a 128GB disk space with only 10-15 satellite passes (1,1/2 Day).  The baseband files are each 8GB-10GB in size.  Is there a way to have SatDump compress these files to 1GB or less. At this rate you can't let SatDump run for more than a day it will require 1TB of disk space or more to run for a few days. Better file compression is need to allow users to keep SatDump running unattended for a long time.  

Thank You

alsldlflglhljlk, 2024-08-01:
I'm not sure why you're getting baseband files with live processing. Live processing should only save the images / output data.

ve3elb, 2024-08-02:
Am not sure but I do. I have installed SatDump on UbuntuOS and PiOS with the same results. Here's from the PiOS with 3 satellite passes this evening.  As you can see the file sizes are 8.xGB each. There should be an option to auto delete baseband files after processing.  This weekend I will have to setup a crontab to delete these file every night so I can keep this setup running unattended.  

[image]

JVital2013, 2024-08-02:
Good morning @ve3elb!

It sounds like you are using autotrack with record selected. If you are only interested in saving the processed images, you can uncheck "record" and only have "live" checked. If you do this, the baseband will not be recorded at all.

If you do want to keep the basebands for some time, keep record checked, but set the IQ decimation. If your SDR is set to 2.4 MHz, a decimation of 10 should work, and still safely capture APT or LRPT.

In the near future we do plan to add a "baseband type" selection, so you can set this to ZIQ, cs8, etc so basebands use even less space.

ve3elb, 2024-08-08:
@JVital2013   thank you for the info. Yes I did have record enabled. It's now disabled and working fine.  

Again Thank You.

closed 2024-08-08 by ve3elb