SatDump/SatDump #769: Could not enum Aaronia Devices!

markste-in, 2024-08-03:
**Description of the issue**
I try to use the Aaronia Spectran V6 ECO with Satdump on Windows 10. The Satdump itself starts and the Spectran is already preselected. If I click play I get a popup saying 
`"Could not enum Aaronia Devices! => D:\a\SatDump\SatDump\plugins\sdr_sources\aaronia_sdr_support\aaronia_sdr.cpp:155"`

The Aaronia RTSA Suite is installed an the AaroniaRTSAAPI.dll file is copied from the sdk folder to the main Aaroia RTSA folder. That seems to be fine since it can find it and loaded the Aaronia API.
**Hardware (SDR/PC/OS)**
Aaronia Spectran V6 ECO
**Version (Eg, 1.0.0, CI Build 171)**
Tried latest release (1.2.0) and the nightly build from August 3rd 2024
**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
```log
[10:55:28 - 03/08/2024] (E) Usage : C:\Program Files\SatDump\bin\satdump-ui.exe [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[10:55:28 - 03/08/2024] (E) Extra options (examples. Any parameter used in modules can be used here) :
[10:55:28 - 03/08/2024] (E)  --samplerate [baseband_samplerate] --baseband_format [f32/i16/i8/w8] --dc_block --iq_swap
[10:55:28 - 03/08/2024] (D) Starting with OpenGL 4.6.0 NVIDIA 392.68
[10:55:28 - 03/08/2024] (D) Max texture size: 16384
[10:55:28 - 03/08/2024] (I)    _____       __  ____                      
[10:55:28 - 03/08/2024] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[10:55:28 - 03/08/2024] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[10:55:28 - 03/08/2024] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[10:55:28 - 03/08/2024] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[10:55:28 - 03/08/2024] (I)                                     /_/      
[10:55:28 - 03/08/2024] (I) Starting SatDump v1.2.0
[10:55:28 - 03/08/2024] (I) 
[10:55:28 - 03/08/2024] (I) Loading config C:\Program Files\SatDump\share\satdump\satdump_cfg.json
[10:55:28 - 03/08/2024] (I) Loading user settings C:\Users\felix\AppData\Roaming/satdump/settings.json
[10:55:28 - 03/08/2024] (I) Loading plugins from C:\Program Files\SatDump\lib\satdump\plugins
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\aaronia_sdr_support.dll...
[10:55:28 - 03/08/2024] (I) Loaded Aaronia API!
[10:55:28 - 03/08/2024] (T) Plugin aaronia_sdr_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\aim_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin aim_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\airspyhf_sdr_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin airspyhf_sdr_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\airspy_sdr_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin airspy_sdr_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\analog_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin analog_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\bladerf_sdr_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin bladerf_sdr_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\bluewalker3_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin bluewalker3_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\cluster_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin cluster_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\cpp_compos_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin standard_cpp_compos loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\cubesat_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin cubesat_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\dmsp_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin dmsp_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\dscovr_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin dscovr_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\dvb_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin dvb_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\elektro_arktika_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin elektro_arktika_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\eos_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin eos_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\fengyun2_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin fengyun2_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\fengyun3_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin fengyun3_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\fengyun4_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin fenyun4_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\gcom_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin gcom_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\geonetcast_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin geonetcast_support loaded!
[10:55:28 - 03/08/2024] (T) Loading plugin C:\Program Files\SatDump\lib\satdump\plugins\gk2a_support.dll...
[10:55:28 - 03/08/2024] (T) Plugin gk2a_support loaded!
[... 523 lines cut ...]
[10:55:29 - 03/08/2024] (D)  - Quadro K2100M
[10:55:29 - 03/08/2024] (I) Loading TLEs from C:\Users\felix\AppData\Roaming/satdump/satdump_tles.txt
[10:55:29 - 03/08/2024] (I) 10062 TLEs loaded!
[10:55:31 - 03/08/2024] (D) Device Spectran V6 000000D776AFDFD8
[10:55:31 - 03/08/2024] (D) Device File Source
[10:55:31 - 03/08/2024] (D) Device Network Source
[10:55:31 - 03/08/2024] (D) Device PlutoSDR IP
[10:55:31 - 03/08/2024] (D) Device RTL-TCP
[10:55:31 - 03/08/2024] (D) Device SDR++ Server
[10:55:31 - 03/08/2024] (D) Device SpyServer
[10:55:31 - 03/08/2024] (T) FFT Rate 8333, Samplerate 1000000, Final Size 8192, Skip 141
[10:56:48 - 03/08/2024] (E) Could not enum Aaronia Devices! => D:\a\SatDump\SatDump\plugins\sdr_sources\aaronia_sdr_support\aaronia_sdr.cpp:155
[11:42:39 - 03/08/2024] (I) Saving user config at C:\Users\felix\AppData\Roaming/satdump/settings.json
[11:42:39 - 03/08/2024] (I) Waiting for the thread...
[11:42:39 - 03/08/2024] (I) Thread stopped
[11:42:39 - 03/08/2024] (I) UI Exit
[11:42:40 - 03/08/2024] (I) Exiting!
```
**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

What I find most confusing is the appearing path
`D:\a\SatDump\SatDump\plugins\sdr_sources\aaronia_sdr_support\aaronia_sdr.cpp:155`
I don't have a D Drive!

Maybe I am just doing it wrong?

AlexandreRouma, 2024-08-03:
It's a known issue with the latest version of RTSASuite (the API used for the device). The Aaronia developers have let me know they will likely push a fix in the coming week.

markste-in, 2024-08-03:
Ah good to know. Do you know if there is any way to download builds before the latest one somewhere from Aaronia. I didn't find a way to dowload older versions to continue some testing?

markste-in, 2024-08-20:
Just wanted to give an update on the problem. Since Aaronia released 3.0.1 build 15665 I compiled the latest nightly build from today (August 20th 2024) with the following following options
`cmake --trace -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/usr -DPLUGIN_AARONIA_SDR_SUPPORT=ON ..
`

When I start Satdump, it is seeing the Aaronia but as soon as I click "start" it is  doing nothing. I can't stop it anymore either? There is no obivious error anymore. Nothing in the log an no popup with an error message.

Here is the log from Satdump:

```log
user@linux-laptop:~/Downloads/SatDump-master/build$ ./satdump-ui
[20:34:51 - 20/08/2024] (E) Usage : ./satdump-ui [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[20:34:51 - 20/08/2024] (E) Extra options (examples. Any parameter used in modules can be used here) :
[20:34:51 - 20/08/2024] (E)  --samplerate [baseband_samplerate] --baseband_format [cf32/cs16/cs8/wav16] --dc_block --iq_swap
[20:34:51 - 20/08/2024] (D) Starting with OpenGL 4.6 (Compatibility Profile) Mesa 24.0.9-0ubuntu0.1
[20:34:51 - 20/08/2024] (D) Max texture size: 16384
[20:34:51 - 20/08/2024] (I)    _____       __  ____                      
[20:34:51 - 20/08/2024] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[20:34:51 - 20/08/2024] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[20:34:51 - 20/08/2024] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[20:34:51 - 20/08/2024] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[20:34:51 - 20/08/2024] (I)                                     /_/      
[20:34:51 - 20/08/2024] (I) Starting SatDump v1.2.1
[20:34:51 - 20/08/2024] (I) 
[20:34:51 - 20/08/2024] (I) Loading config satdump_cfg.json
[20:34:51 - 20/08/2024] (I) Loading user settings /home/user/.config/satdump/settings.json
[20:34:51 - 20/08/2024] (I) Loading plugins from ./plugins
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libgk2a_support.so...
[20:34:51 - 20/08/2024] (T) Plugin gk2a_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/liborbcomm_support.so...
[20:34:51 - 20/08/2024] (T) Plugin orbcomm_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libmirisdr_sdr_support.so...
[20:34:51 - 20/08/2024] (T) Plugin mirisdr_sdr_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libairspyhf_sdr_support.so...
[20:34:51 - 20/08/2024] (T) Plugin airspyhf_sdr_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libdvb_support.so...
[20:34:51 - 20/08/2024] (T) Plugin dvb_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libsimd_sse41.so...
[20:34:51 - 20/08/2024] (T) Plugin simd_sse41 loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libremote_sdr_support.so...
[20:34:51 - 20/08/2024] (T) Plugin remote_sdr_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libtubsat_support.so...
[20:34:51 - 20/08/2024] (T) Plugin tubsat_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libspacex_support.so...
[20:34:51 - 20/08/2024] (T) Plugin spacex_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libelektro_arktika_support.so...
[20:34:51 - 20/08/2024] (T) Plugin elektro_arktika_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libfengyun2_support.so...
[20:34:51 - 20/08/2024] (T) Plugin fengyun2_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libsdrpp_server_support.so...
[20:34:51 - 20/08/2024] (T) Plugin spyserver_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libeos_support.so...
[20:34:51 - 20/08/2024] (T) Plugin eos_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libstereo_support.so...
[20:34:51 - 20/08/2024] (T) Plugin stereo_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/liblandsat_support.so...
[20:34:51 - 20/08/2024] (T) Plugin landsat_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/liboceansat_support.so...
[20:34:51 - 20/08/2024] (T) Plugin oceansat_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libdmsp_support.so...
[20:34:51 - 20/08/2024] (T) Plugin dmsp_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libaim_support.so...
[20:34:51 - 20/08/2024] (T) Plugin aim_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libwsf_support.so...
[20:34:51 - 20/08/2024] (T) Plugin wsf_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libhinode_support.so...
[20:34:51 - 20/08/2024] (T) Plugin hinode_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libmats_support.so...
[20:34:51 - 20/08/2024] (T) Plugin mats_support loaded!
[20:34:51 - 20/08/2024] (T) Loading plugin ./plugins/libothers_support.so...
[20:34:51 - 20/08/2024] (T) Plugin others_support loaded!
[... 523 lines cut ...]
[20:34:51 - 20/08/2024] (D)  - Intel(R) UHD Graphics 630
[20:34:51 - 20/08/2024] (D)  - NVIDIA GeForce RTX 2080 with Max-Q Design
[20:34:51 - 20/08/2024] (I) Loading TLEs from /home/user/.config/satdump/satdump_tles.txt
[20:34:51 - 20/08/2024] (I) 10127 TLEs loaded!
[20:34:51 - 20/08/2024] (D) Device Spectran V6 0x7ffcf11abd98
[20:34:51 - 20/08/2024] (D) Device File Source
[20:34:51 - 20/08/2024] (D) Device Network Source
[20:34:51 - 20/08/2024] (D) Device RTL-TCP
[20:34:51 - 20/08/2024] (D) Device SDR++ Server
[20:34:51 - 20/08/2024] (D) Device SpyServer
[... cut ...]
[20:35:09 - 20/08/2024] (I) Set Spectran decimation to 2
[20:35:09 - 20/08/2024] (D) Set Aaronia frequency to 97574000
[20:35:09 - 20/08/2024] (D) Set Aaronia USB compression mode to 0
[20:35:09 - 20/08/2024] (D) Set Aaronia reflevel to -20.000000
[20:35:09 - 20/08/2024] (D) Set Aaronia AGC mode to 0
[20:35:09 - 20/08/2024] (I) Waiting for device to stream...
[20:35:09 - 20/08/2024] (I) Started Aaronia Device!
[20:35:09 - 20/08/2024] (T) FFT Rate 434716, Samplerate 46080000, Final Size 16384, Skip 418332
```

The OS is an Ubuntu 24.04 LTS

markste-in, 2024-08-21:
I did some troubleshooting and build Satdump with some debug / log statements. I also added a sample rate of 46080000 to aaronia_sdr.cpp since this is the one supported by the Spectran V6 ECO.

It seems it always get stuck at (aaronia_sdr.h at line 59)

```cpp
while ((res = AARTSAAPI_GetPacket(&aaronia_device, 0, 0, &packet)) == AARTSAAPI_EMPTY)
                usleep(1000);
logger->info("Got  non-empty package with code %x", res);
```

It  is either always AARTSAAPI_EMPTY or i get 0x8000000a (AARTSAAPI_ERROR_INVALID_SIZE).

I am not sure why. Is there any kind of documentation about the Aaronia SDK?
I guess it never really started streaming to begin with. 

```log
[11:44:04 - 21/08/2024] (E) SatDump has NOT been built in Release mode.
[11:44:04 - 21/08/2024] (E) If you are not a developer but intending to use the software,
[11:44:04 - 21/08/2024] (E) you probably do not want this. If so, make sure to
[11:44:04 - 21/08/2024] (E) specify -DCMAKE_BUILD_TYPE=Release in CMake.
[11:44:04 - 21/08/2024] (D) Device Spectran V6 0x7ffec646f288
[11:44:04 - 21/08/2024] (D) Device File Source
[11:44:04 - 21/08/2024] (D) Device Network Source
[11:44:04 - 21/08/2024] (D) Device RTL-TCP
[11:44:04 - 21/08/2024] (D) Device SDR++ Server
[11:44:04 - 21/08/2024] (D) Device SpyServer
[11:44:04 - 21/08/2024] (T) FFT Rate 8333, Samplerate 1000000, Final Size 8192, Skip 141
[11:44:15 - 21/08/2024] (I) Set Spectran receiver clock to 46MHz
[11:44:15 - 21/08/2024] (I) Set Spectran decimation to 1
[11:44:15 - 21/08/2024] (D) Set Aaronia frequency to 97574000
[11:44:15 - 21/08/2024] (D) Set Aaronia USB compression mode to 0
[11:44:15 - 21/08/2024] (D) Set Aaronia reflevel to -20.000000
[11:44:15 - 21/08/2024] (D) Set Aaronia AGC mode to 0
[11:44:15 - 21/08/2024] (I) Get device state: 0x10000002
[11:44:15 - 21/08/2024] (I) Get available packages: 0
[11:44:15 - 21/08/2024] (I) Waiting for device to stream...
[11:44:15 - 21/08/2024] (I) Consuming first package...
[11:44:15 - 21/08/2024] (I) Started Aaronia Device!
[11:44:15 - 21/08/2024] (I) Worker thread initialised
[11:44:15 - 21/08/2024] (T) FFT Rate 434716, Samplerate 46080000, Final Size 16384, Skip 418332
[11:44:15 - 21/08/2024] (I) Inside mainThread ...
[11:44:15 - 21/08/2024] (I) Get device state: 0x10000002
[11:44:15 - 21/08/2024] (I) Get available packages: 0
[11:44:15 - 21/08/2024] (I) Waiting for non-empty package ...
[11:44:15 - 21/08/2024] (I) Got  non-empty package with code 8000000a
[11:44:15 - 21/08/2024] (I) Get device state: 0x10000002
[11:44:15 - 21/08/2024] (I) Get available packages: 0
[11:44:15 - 21/08/2024] (I) Waiting for non-empty package ...
```

markste-in, 2024-08-21:
further little update
After starting the device in `AaroniaSource::start()` with
```cpp
    if (AARTSAAPI_StartDevice(&aaronia_device) != AARTSAAPI_OK)
        throw satdump_exception("Could not start Aaronia device!");
```
I added the following code to check if the device goes into running mode.

```cpp
    logger->info("Waiting for device state AARTSAAPI_RUNNING");
    while (AARTSAAPI_GetDeviceState(&aaronia_device) != AARTSAAPI_RUNNING)
    {
#ifdef _WIN32
        Sleep(100);
#else
        usleep(100000);
#endif
        logger->info("Still waiting");
    }
```

unfortunately it never does... 
My current assumption is that the device is configured wrong? Maybe?

markste-in, 2024-11-06:
> There's a known issue with the latest version of RTSASuite (the API used for the device) that causes the RTSAAPI initialization to fail. The Aaronia developers have let me know they will likely push a fix in the coming week.

Do you know about any update from Aaronia?

markste-in, 2024-11-12:
I got the ECO version of the Aaronia Spectran V6 working. Those changes conflict with the more expensive Plus model at the moment and need a little bit of rework. Unfortunately I couldn't test any of the refactoring work needed since I don't have / can't afford the Plus version.
If people want to test Satdump with the ECO they can still use my current fork

https://github.com/markste-in/SatDump

markste-in, 2025-06-30:
Fixed with https://github.com/SatDump/SatDump/pull/862

closed 2025-06-30 by markste-in