SatDump/SatDump #771: Debs for newer debian-like distros

pkreuzt, 2024-08-03:
Current released debs work fine on stable Debian/PiOS/Ubuntu, but there are a lot of derivatives based on Testing/Sid branches (as generally speaking packages for testing and sid are compatible with each other).
I am myself running Debian Testing on Rpi just because of the known USB-blocking bug of older kernels (I suppose it may be fixed now, but not sure). And every time changes to source are made I have to rebuild SatDump deb locally, as nightly debs depend on way older software versions.

Would it be too much a hassle to release additional debs for these versions?

pkreuzt, 2024-08-05:
> That likely won't happen, since Unstable is well, unstable, and neither Testing nor Unstable are covered by Debian Security. I'd recommend making a `systemd` unit or `cron` job to recompile SatDump once every week or so.

For amd64, newer Ubuntu packages should work on Debian Testing/Sid. I don't use Ubuntu, but sometimes read complains about specific packages being older versions than the ones available on Testing (Ubuntu suposedly uses Testing as baseline). Tried the 24.04 package in my Deb Trixie and seems to work, so maybe Satdump dependencies are not affected by that policy.

But arm64 case is different. Only Stable/RaspiOS package is available here. It is not compatible with any Debian derivative wich is not Stable. I bet not even with Ubuntu.

alsldlflglhljlk, 2024-08-08:
Well, Ubuntu does use Testing as a base, but Ubuntu is Ubuntu and its packages may depend on _slightly_ different library versions than Debian Stable. An Ubuntu package will only work reliably on Ubuntu and a Debian package will only work reliably on Debian. You might get lucky with an Ubuntu package working on Debian, but that's otherwise known as [FrankenDebian](https://wiki.debian.org/DontBreakDebian) and that's not good.

JVital2013, 2024-08-20:
This likely won't come right away, but there probably will be some more ARM builds in the future since ARM platforms are proliferating.

As usual, though, with pre-built packages it can be tricky to ensure the right versions are targeted for different platforms, and there's the matter of configuring a runner (or cross-compellation) runner to appropriately build the packages. All doable - but in due time!

pkreuzt, 2024-08-20:
I'm not a dev, just launching some ideas. I reckon software with that specific dependencies is difficult to maintain for a wide array of platforms and that it makes sense to keep release methods as simple as possible.

Some projects manage to build somewhat generic deb packages targeted for Debian Stable / older Ubuntu LTS on one side plus Testing / Sid / newer Ubuntu on the other. Another alternative could be pack most specific dependencies inside and make an out-of-tree install (i.e. to /opt). This would make packages bigger but also more portable.
Also I think it makes sense for ARM builds to split GUI component apart, as it is barely usable on these platforms.