SatDump/SatDump #773: How do I fix "Error: SPS is invalid. Must be above 1!"

jmiller2015, 2024-08-06:
Hello,

I am just getting started with SDR, and want to view weather satellite images. I am working with a nooelec SMArt v5, and a home-built QFH antenna.

I have been fiddling with the SatDump settings and have a good signal from GOES-14, but am unable to process the data. I have been getting the error in the title of this post whenever I try to process the live data.

I have attached a screenshot of my view in the SatDump program. Can someone walk me through how to set up the live data processing? I'm still learning the program, and have watched several tutorial videos, but nothing has worked so far. I have SatDump set up the same way as this video: https://www.youtube.com/watch?v=XMDAiUjzkhw&t=157s.

[image]
[image]

alsldlflglhljlk, 2024-08-06:
Two things:

1 - The higher screenshot shows you are using a decimation of 10. Decimation skips samples and lowers the effective sample rate. You really don't need decimation for GOES, so set decimation to 1.

2 - You are using a sample rate of 3 Msps, but the Nooelec SMArt v5 is based on the RTL2832 ADC which can only sample up to about ≈ 2.56 Msps reliably. The bandwidth of GOES is only 1.205 MHz, so lowering the sample rate should help.

alsldlflglhljlk, 2024-08-06:
Additionally, a QFH won't get you the gain you need to receiving GOES. If you're just getting started with SDR, I'd highly recommend receiving NOAA APT or even Meteor-M LRPT before moving into the L-Band. @sgcderek has a [really good video series](https://www.youtube.com/watch?v=-NvyEIuoLnc&list=PL2kdEU4_yN-ebVoRsnYBILgPgcQNjjZ0z) on this topic...

jmiller2015, 2024-08-06:
Adjusting the sample rate allowed me to see the BPSK modulator and the other data with it. Thank you!

-------- Original Message --------

alsldlflglhljlk, 2024-08-08:
Glad to hear you solved your problem! Can we close this, @JVital2013?

JVital2013, 2024-08-09:
Sounds good to me - thanks for chipping in everyone!

closed 2024-08-09 by JVital2013