SatDump/SatDump #775: Unable to use a spyserver source for satdump autotrack

krdee1, 2024-08-10:
I've been trying and failing to use `satdump autotrack` with a spyserver device. I suspect the connection is not being made to the spyserver due to a misconfiguration on my part, but I can't find it. I only see connections appear in my spyserver logs when I connect from SDR++, but never when I start autotrack.

To briefly describe the hardware situation, I have a RTL-SDR.com dongle (v1) on a raspberry pi 3b+ wirelessly networked onto a LAN with a linux server. The raspberry pi is running spyserver, which I can confirm works flawlessly with a SDR++ client on the same LAN. The linux server is intended to run `satdump autotrack` with the raspberry pi spyserver configured as the source device. The satdump version is 1.2.0-1.

The autotrack web console shows the device inactive during a pass, and the FFT is a flat line, but everything else appears to be correct.

[image] 

The config I pass in with `satdump autotrack` is as follows (with some anonymization, of course).

```
{
    "parameters": {
        "source": "spyserver",
        "ip_address": "192.168.7.251",
        "samplerate": 2048000,
        "port": "5555",
        "bit_depth": 32,
        "gain": 0,
        "digital_gain": 0,
        "initial_frequency": 137.64e6,
        "fft_enable": true,
        "fft_size": 65536, 
        "fft_rate": 30,
        "fft_avgn": 1
    },
    "finish_processing": true,
    "output_folder": "/srv/output",
    "qth": {
        "lat": XX.XXXX,
        "lon": YY.YYYY,
        "alt": Z
    },
    "http_server": "192.168.1.92:8081",
    "tracked_objects": [
        {
            "norad": 25338,
            "downlinks": [
                {
                    "frequency": 137620000,
                    "record": false,
                    "live": true,
                    "pipeline_name": "noaa_apt",
                    "pipeline_params": {
                        "autocrop_wedges": false,
                        "satellite_number": "15",
                        "sdrpp_noise_reduction": true
                    }
                 },
                 {
                     "frequency": 137350000,
                     "record": false,
                     "live": true,
                     "pipeline_name": "noaa_dsb",
                     "pipeline_params": {}
                }
            ]
        },
        {
            "norad": 28654,
            "downlinks": [
                {
                    "frequency": 137912500,
                    "record": false,
                    "live": true,
                    "pipeline_name": "noaa_apt",
                    "pipeline_params": {
                        "autocrop_wedges": false,
                        "satellite_number": "18",
                        "sdrpp_noise_reduction": true
                    }
                 },
                 {
                     "frequency": 137350000,
                     "record": false,
                     "live": true,
                     "pipeline_name": "noaa_dsb",
                     "pipeline_params": {}
                }
            ]
        },
        {
            "norad": 33591,
            "downlinks": [
                {
                    "frequency": 137100000,
                    "record": false,
                    "live": true,
                    "pipeline_name": "noaa_apt",
                    "pipeline_params": {
                        "autocrop_wedges": false,
                        "satellite_number": "19",
                        "sdrpp_noise_reduction": true
                    }
                 },
                {
                     "frequency": 137770000,
                     "record": false,
                     "live": true,
                     "pipeline_name": "noaa_dsb",
                     "pipeline_params": {}
                }
            ]
        },
        {
            "norad": 57166,
            "downlinks": [
                {
                    "frequency": 137900000,
                    "record": false,
                    "live": true,
                    "pipeline_name": "meteor_m2-x_lrpt",
                    "pipeline_params": {
                        "dc_block": true
                    }
                }
            ]
        }
    ]
}
```

And, just in case it helps, the spyserver config:

```
# SPY Server Configuration File

# TCP Listener
#
bind_host = 192.168.7.251
bind_port = 5555-6666

# List Server in Airspy Directory
#
list_in_directory = 0

# User sessions
#
maximum_clients = 3

# Maximum session duration
# In minutes. 0 for no limit
#
#maximum_session_duration = 30

# Allow clients to retune and change of gain of the device
#
allow_control = 1

# Device Type
# Possible Values:
#   AirspyOne (R0, R2, Mini)
#   AirspyHF+
#   RTL-SDR
#   Auto      (Scans for the first available device)
#
device_type = RTL-SDR

# Device Serial Number as 64bit Hex
# For example: 0xDD52D95C904534AD
# A value of 0 will acquire the first available device.
#
device_serial = 0

# Device Sample Rate
# Possible Values:
#   Airspy R0, R2 : 10000000 or 2500000
#   Airspy Mini   :  6000000 or 3000000
#   Airspy HF+    :   768000
#   RTL-SDR       :   500000 to 3200000
# Comment to use the device's default
#
device_sample_rate = 2048000

# Force 8bit Compression Mode
# The 8bit Compression mode has proven sufficiently good for most streaming use cases.
# Use it to same some internet bandwidth.
#
#force_8bit = 1

# Maximum Bandwidth
# Limits the maximum IQ bandwidth the clients can set
# Recommended value for WFM is 200000
# Recommended value for narrow band modes is 15000
#
maximum_bandwidth = 200000

# FFT Frames Per Second
#
fft_fps = 15

# FFT Bins
# Bins = 2^fft_bin_bits
#
fft_bin_bits = 16

# Initial Center Frequency
#
initial_frequency = 137543210

# Minimum Tunable Frequency
# Comment if using the device default
#
#minimum_frequency = 0

# Maximum Tunable Frequency
# Comment if using the device default
#
#maximum_frequency = 35000000

# Frequency Correction in PPB
#
#frequency_correction_ppb = 0

# Initial Gain
#
initial_gain = 15

# RTL-SDR Sampling mode
# Quadrature = 0,
# Direct Sampling I Branch = 1
# Direct Sampling Q Branch = 2
#rtl_sampling_mode = 0

# Converter Offset
# Set to -120000000 to enable the SpyVerter offset
#converter_offset = -120000000

# Bias-Tee
# For AirspyOne only - Useful for LNA's and SpyVerter
#enable_bias_tee = 0

# Buffer Size (in milliseconds)
#
buffer_size_ms = 50
```

alsldlflglhljlk, 2024-08-10:
Try connecting to the SpyServer with the GUI via the Recorder tab. Does that work?

krdee1, 2024-08-11:
I was able to get it working on my desktop (from which I was testing with SDR++) but I can't easily test the same way on the server since it's a headless setup. I did notice that I can't ping the spyserver raspi from the satdump server, even though I can in the other direction, so I think this is just a network issue. I'll continue trying to work it out.

krdee1, 2024-08-11:
Thanks for the debugging tip, I had the spyserver and satdump server on different subnets, so I had to explicitly configure a route for them to be able to communicate with each other. Once I had that set up, I was able to get results.

[image]

closed 2024-08-11 by krdee1