SatDump/SatDump #776: (nightly) CLI recording at exactly 4 MHz records roughly half the requested data (saving 4 MHz sampling takes more time)

tstellanova, 2024-08-10:
**Description of the issue**
I'm using the CLI to script recording GPS L2C from Pluto+ SDR (over IP)
If I record at a sample rate of exactly 4 MHz, I find that only about half the expected data is saved to disk.
If I record at  4 +/- 1 MHz I find the expected data is written to disk. 

My goal is to record at least 10 seconds worth of `cs8` data. I use a command line of the form:
`/Applications/SatDump.app/Contents/MacOS/satdump record ./data/pluto_gnss_L2C_10s_0001 --source plutosdr --frequency 1227600000 --samplerate 4000000 --baseband_format cs8  --gain_mode 3 --gain 40 --ip_address 10.0.1.191 --auto_reconnect true --timeout 11 `

**Hardware (SDR/PC/OS)**

Pluto+ SDR (over IP)
macos 14.5 sonoma 
macbook pro / apple m3 max

**Version (Eg, 1.0.0, CI Build 171)**

SatDump v1.2.1-b7cbaae (from nightly release)

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
 Does not crash.

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

Partial log from 4 MHz sampling rate:

```
/Applications/SatDump.app/Contents/MacOS/satdump record ./data/pluto_gnss_L2C_10s_0001 --source plutosdr --frequency 1227600000 --samplerate 4000000 --baseband_format cs8  --gain_mode 3 --gain 40 --ip_address 10.0.1.191 --auto_reconnect true --timeout 11 
...
[19:40:04 - 10/08/2024] (T) Using PlutoSDR IP Address 10.0.1.191
[19:40:04 - 10/08/2024] (D) Set PlutoSDR samplerate to 4000000
[19:40:05 - 10/08/2024] (D) Set PlutoSDR frequency to 1227600000
[19:40:05 - 10/08/2024] (D) Set PlutoSDR gain to 40, mode hybrid
[19:40:05 - 10/08/2024] (T) PlutoSDR stream with 4 buffers of size 133333
[19:40:05 - 10/08/2024] (I) Wrote 0 MB
[19:40:05 - 10/08/2024] (I) Wrote 0 MB
[19:40:05 - 10/08/2024] (I) Wrote 0 MB
[19:40:05 - 10/08/2024] (I) Wrote 1 MB
[19:40:07 - 10/08/2024] (I) Wrote 5 MB
[19:40:07 - 10/08/2024] (I) Wrote 5 MB
[19:40:07 - 10/08/2024] (I) Wrote 6 MB
[19:40:07 - 10/08/2024] (I) Wrote 6 MB
[19:40:07 - 10/08/2024] (I) Wrote 6 MB
[19:40:07 - 10/08/2024] (I) Wrote 7 MB
[19:40:07 - 10/08/2024] (I) Wrote 7 MB
[19:40:07 - 10/08/2024] (I) Wrote 7 MB
[19:40:07 - 10/08/2024] (I) Wrote 8 MB
[19:40:07 - 10/08/2024] (I) Wrote 8 MB
[19:40:09 - 10/08/2024] (I) Wrote 13 MB
[19:40:09 - 10/08/2024] (I) Wrote 13 MB
[19:40:09 - 10/08/2024] (I) Wrote 14 MB
[19:40:09 - 10/08/2024] (I) Wrote 14 MB
[19:40:09 - 10/08/2024] (I) Wrote 15 MB
[19:40:09 - 10/08/2024] (I) Wrote 15 MB
[19:40:09 - 10/08/2024] (I) Wrote 15 MB
[19:40:09 - 10/08/2024] (I) Wrote 16 MB
[19:40:09 - 10/08/2024] (I) Wrote 16 MB
[19:40:11 - 10/08/2024] (I) Wrote 21 MB
[19:40:11 - 10/08/2024] (I) Wrote 21 MB
[19:40:11 - 10/08/2024] (I) Wrote 22 MB
[19:40:11 - 10/08/2024] (I) Wrote 22 MB
[19:40:11 - 10/08/2024] (I) Wrote 22 MB
[19:40:11 - 10/08/2024] (I) Wrote 23 MB
[19:40:11 - 10/08/2024] (I) Wrote 23 MB
[19:40:11 - 10/08/2024] (I) Wrote 24 MB
[19:40:11 - 10/08/2024] (I) Wrote 24 MB
[19:40:13 - 10/08/2024] (I) Wrote 29 MB
[19:40:13 - 10/08/2024] (I) Wrote 29 MB
[19:40:13 - 10/08/2024] (I) Wrote 30 MB
[19:40:13 - 10/08/2024] (I) Wrote 30 MB
[19:40:13 - 10/08/2024] (I) Wrote 30 MB
[19:40:13 - 10/08/2024] (I) Wrote 31 MB
[19:40:13 - 10/08/2024] (I) Wrote 31 MB
[19:40:13 - 10/08/2024] (I) Wrote 31 MB
[19:40:13 - 10/08/2024] (I) Wrote 32 MB
[19:40:13 - 10/08/2024] (I) Wrote 32 MB
[19:40:15 - 10/08/2024] (I) Wrote 37 MB
[19:40:15 - 10/08/2024] (I) Wrote 37 MB
[19:40:15 - 10/08/2024] (I) Wrote 38 MB
[19:40:15 - 10/08/2024] (I) Wrote 38 MB
[19:40:15 - 10/08/2024] (I) Wrote 39 MB
[19:40:15 - 10/08/2024] (I) Wrote 39 MB
[19:40:15 - 10/08/2024] (I) Wrote 39 MB
[19:40:15 - 10/08/2024] (I) Wrote 40 MB
[19:40:15 - 10/08/2024] (I) Wrote 40 MB
[19:40:16 - 10/08/2024] (W) Timeout is over! (11s >= 11s) Stopping.
[19:40:16 - 10/08/2024] (I) Waiting for the thread...
[19:40:16 - 10/08/2024] (I) Thread stopped
[19:40:16 - 10/08/2024] (I) Waiting for the thread...
[19:40:16 - 10/08/2024] (I) Thread stopped
[19:40:16 - 10/08/2024] (I) Done! Goodbye
(btl_env) todd@relap sigsplat % ls -latr ./data/pluto_gnss_L2C*
-rwx------  1 todd  staff  41066564 Aug 10 12:40 ./data/pluto_gnss_L2C_10s_0001.cs8
```
This is about half the expected data size of 80000000. It's notable that the timestamp for `[19:40:16 - 10/08/2024] (W) Timeout is over! (11s >= 11s) Stopping.` log line is indeed about 10 seconds after the `[19:40:05 - 10/08/2024] (I) Wrote 0 MB` log line. 

For comparison, partial log from 4.1 MHz sampling:

```
/Applications/SatDump.app/Contents/MacOS/satdump record ./data/pluto_gnss_L2C_10s_0001 --source plutosdr --frequency 1227600000 --samplerate 4100000 --baseband_format cs8  --gain_mode 3 --gain 40 --ip_address 10.0.1.191 --auto_reconnect true --timeout 11 
...
[20:12:40 - 10/08/2024] (I) Starting SatDump v1.2.1-b7cbaae
...
[20:12:41 - 10/08/2024] (T) Using PlutoSDR IP Address 10.0.1.191
[20:12:41 - 10/08/2024] (D) Set PlutoSDR samplerate to 4100000
[20:12:42 - 10/08/2024] (D) Set PlutoSDR frequency to 1227600000
[20:12:42 - 10/08/2024] (D) Set PlutoSDR gain to 40, mode hybrid
[20:12:42 - 10/08/2024] (T) PlutoSDR stream with 4 buffers of size 136666
[20:12:42 - 10/08/2024] (I) Wrote 0 MB
[20:12:42 - 10/08/2024] (I) Wrote 0 MB
[20:12:42 - 10/08/2024] (I) Wrote 0 MB
[20:12:42 - 10/08/2024] (I) Wrote 1 MB
[20:12:42 - 10/08/2024] (I) Wrote 2 MB
[20:12:42 - 10/08/2024] (I) Wrote 3 MB
[20:12:42 - 10/08/2024] (I) Wrote 4 MB
[20:12:42 - 10/08/2024] (I) Wrote 5 MB
[20:12:44 - 10/08/2024] (I) Wrote 14 MB
[20:12:44 - 10/08/2024] (I) Wrote 15 MB
[20:12:44 - 10/08/2024] (I) Wrote 16 MB
[20:12:44 - 10/08/2024] (I) Wrote 17 MB
[20:12:44 - 10/08/2024] (I) Wrote 18 MB
[20:12:44 - 10/08/2024] (I) Wrote 19 MB
[20:12:44 - 10/08/2024] (I) Wrote 19 MB
[20:12:44 - 10/08/2024] (I) Wrote 20 MB
[20:12:44 - 10/08/2024] (I) Wrote 21 MB
[20:12:44 - 10/08/2024] (I) Wrote 21 MB
[20:12:45 - 10/08/2024] (W) PlutoSDR underflow!
[20:12:45 - 10/08/2024] (W) PlutoSDR underflow!
[20:12:46 - 10/08/2024] (I) Wrote 30 MB
[20:12:46 - 10/08/2024] (I) Wrote 31 MB
[20:12:46 - 10/08/2024] (I) Wrote 32 MB
[20:12:46 - 10/08/2024] (I) Wrote 33 MB
[20:12:46 - 10/08/2024] (I) Wrote 34 MB
[20:12:46 - 10/08/2024] (I) Wrote 34 MB
[20:12:46 - 10/08/2024] (I) Wrote 35 MB
[20:12:46 - 10/08/2024] (I) Wrote 36 MB
[20:12:46 - 10/08/2024] (I) Wrote 37 MB
[20:12:46 - 10/08/2024] (I) Wrote 38 MB
[20:12:48 - 10/08/2024] (I) Wrote 45 MB
[20:12:48 - 10/08/2024] (W) PlutoSDR underflow!
[20:12:48 - 10/08/2024] (I) Wrote 46 MB
[20:12:48 - 10/08/2024] (W) PlutoSDR underflow!
[20:12:48 - 10/08/2024] (I) Wrote 46 MB
[20:12:48 - 10/08/2024] (W) PlutoSDR underflow!
[20:12:48 - 10/08/2024] (I) Wrote 47 MB
[20:12:48 - 10/08/2024] (I) Wrote 48 MB
[20:12:48 - 10/08/2024] (I) Wrote 49 MB
[20:12:48 - 10/08/2024] (I) Wrote 50 MB
[20:12:48 - 10/08/2024] (I) Wrote 51 MB
[20:12:48 - 10/08/2024] (I) Wrote 51 MB
[20:12:50 - 10/08/2024] (I) Wrote 61 MB
[20:12:50 - 10/08/2024] (I) Wrote 62 MB
[20:12:50 - 10/08/2024] (I) Wrote 63 MB
[20:12:50 - 10/08/2024] (I) Wrote 63 MB
[20:12:50 - 10/08/2024] (I) Wrote 64 MB
[20:12:50 - 10/08/2024] (I) Wrote 65 MB
[20:12:50 - 10/08/2024] (I) Wrote 66 MB
[20:12:50 - 10/08/2024] (I) Wrote 67 MB
[20:12:50 - 10/08/2024] (I) Wrote 68 MB
[20:12:50 - 10/08/2024] (I) Wrote 68 MB
[20:12:52 - 10/08/2024] (I) Wrote 77 MB
[20:12:52 - 10/08/2024] (I) Wrote 78 MB
[20:12:52 - 10/08/2024] (I) Wrote 79 MB
[20:12:52 - 10/08/2024] (I) Wrote 80 MB
[20:12:52 - 10/08/2024] (I) Wrote 80 MB
[20:12:52 - 10/08/2024] (I) Wrote 81 MB
[20:12:52 - 10/08/2024] (I) Wrote 82 MB
[20:12:52 - 10/08/2024] (I) Wrote 83 MB
[20:12:52 - 10/08/2024] (I) Wrote 84 MB
[20:12:52 - 10/08/2024] (I) Wrote 85 MB
[20:12:53 - 10/08/2024] (W) Timeout is over! (11s >= 11s) Stopping.
[20:12:53 - 10/08/2024] (I) Waiting for the thread...
[20:12:53 - 10/08/2024] (I) Thread stopped
[20:12:53 - 10/08/2024] (I) Waiting for the thread...
[20:12:53 - 10/08/2024] (I) Thread stopped
[20:12:53 - 10/08/2024] (I) Done! Goodbye
(btl_env) todd@relap sigsplat % ls -latr ./data/pluto_gnss_L2*
-rwx------  1 todd  staff  86099580 Aug 10 13:12 ./data/pluto_gnss_L2C_10s_0001.cs8
```
The size of the data saved in this case is about what I'd expect for 10 seconds * 4.1E6 samples * 2 bytes per `cs8` sample = 82 MB.

Based on the timestamps logged, it appears that, for some reason, saving 4 MHz is taking twice as long as saving  4.1 MHz.

alsldlflglhljlk, 2024-08-10:
Does a sample rate of 4.000001 Msps show the same behavior? Have you tried with a different frequency?

tstellanova, 2024-08-11:
Yes, I have tried with a variety of other frequencies. I’m currently working around the problem by sampling at 3 MHz (which is adequate in this case).  I just now tried with 400000140000104000100 MspsThe two lower sample rates exhibit the same issue. The problem vanishes at 4000100 Msps

alsldlflglhljlk, 2024-08-11:
Does a format other than `cs8` also take unusually long?

tstellanova, 2024-08-11:
I just now tried with `cs16` and `cf32`  output, 4.0 Msps and then 4.000100 Msps and indeed, 4.0 records about half as much data as expected in the allotted time with both output formats.

tstellanova, 2024-08-11:
FWIW I also briefly tried with the `hackrf` source and do not see this problem. This appears to be an issue with the `plutosdr` source.

alsldlflglhljlk, 2024-08-11:
Does recording for 10 seconds in the GUI exhibit the same behavior?

tstellanova, 2024-08-12:
Yes, this issue is also exhibited via the GUI.  I just now tried recording `cs8` at 4.0 Msps and then 4.000100 Msps, for about 10 seconds (manually timing and clicking start/stop in the "Recording" GUI) and this is what satdump recorded:
```
-rw-r--r--@  1 todd  staff    44533222 Aug 12 06:57 2024-08-12_13-57-27_4000000SPS_1575420000Hz.cs8
-rw-r--r--@  1 todd  staff    86135056 Aug 12 06:58 2024-08-12_13-58-26_4000099SPS_1575420000Hz.cs8
```
Again, roughly 2x as much data. I'll update the title of this issue to reflect this.

AaronJackson, 2024-10-20:
I'm wondering if this relates to my issues with receiving NOAA (which I'm new to). Using Pluto SDR at 4Msps at cs32, an overhead pass gives a file of around 5.7GB. Decoding the baseband recording results in this weird streaky output. I will try again tomorrow with a different sample rate

[image]

AaronJackson, 2024-10-20:
Ok not the best pass but at 4.1MS/s it confirms the issue.
i will try and do some digging but I doubt I'll be able to find the issue
[image]