SatDump/SatDump #778: Error during build

FabC, 2024-08-12:
Got several error almost at the end of the build procedure, seems libraries are outdated, report just a couple of them:

/home/****/SatDump/plugins/sdr_sources/usrp_sdr_support/usrp_sdr.cpp:44:58: error: ‘class uhd::usrp::multi_usrp’ has no member named ‘get_master_clock_rate_range’; did you mean ‘get_master_clock_rate’?
   44 |         uhd::meta_range_t dev_samplerates = usrp_device->get_master_clock_rate_range();
      |                                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                          get_master_clock_rate

/home/****/SatDump/plugins/sdr_sources/usrp_sdr_support/usrp_sdr.cpp:44:58: error: ‘class uhd::usrp::multi_usrp’ has no member named ‘get_master_clock_rate_range’; did you mean ‘get_master_clock_rate’?
   44 |         uhd::meta_range_t dev_samplerates = usrp_device->get_master_clock_rate_range();
      |                                                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                          get_master_clock_rate

I followed the procedure in https://github.com/SatDump/SatDump. I'm using Linux Mint 21.3, thank in advance.

Aang23, 2024-08-12:
I believe Mint 21.3 is based on Ubuntu 22.04. If it is so, the built-in UHD is simply too old and I'll have to tell you to manually install/build newer than that. We can't keep supporting long outdated versions because of LTS distros still shipping them...

FabC, 2024-08-12:
Mint 21.3 was released In jan-2024 and yes, it is based on Ubuntu 22.04.
Please let me know the UHD version to consider.

Fabrizio

Aang23, 2024-08-12:
The newest - so I believe 4.6.x at the moment. 2 years old for that kind of package is getting really old...

FabC, 2024-08-15:
Yes, indeed Mint packages the UHD version 4.1 that is dated 2022. I tried
to install the 4.6.0 .deb package but I found unresolved dependencies
issues: any hints for the installation ? Thanks

JVital2013, 2024-08-20:
It might be best to uninstall UHD, then build it yourself from source. The code is available at https://github.com/EttusResearch/uhd, but you only need to build/install within the `host` directory

third-to-lastperson, 2024-09-13:
I encountered an error during compilation. How can I solve it? My platform is Ubuntu 20.04

/home/wsf/SatDump/src-core/common/utils.cpp:89:53: error: ‘CURLSSLOPT_NATIVE_CA’ was not declared in this scope
   89 |         curl_easy_setopt(curl, CURLOPT_SSL_OPTIONS, CURLSSLOPT_NATIVE_CA);
      |                                                     ^~~~~~~~~~~~~~~~~~~~
/home/wsf/SatDump/src-core/common/utils.cpp: In function ‘int perform_http_request_post(std::string, std::string&, std::string)’:
/home/wsf/SatDump/src-core/common/utils.cpp:126:53: error: ‘CURLSSLOPT_NATIVE_CA’ was not declared in this scope
  126 |         curl_easy_setopt(curl, CURLOPT_SSL_OPTIONS, CURLSSLOPT_NATIVE_CA)

JVital2013, 2024-09-29:
@third-to-lastperson this issue has been fixed recently. Update to the latest master branch and it should build for you.

@FabC would you like any further assistance with your issue? For the most part, it's just uninstalling uhd from the repos, and building it from source, but we might be able to provide pointers if you have issues!

closed 2026-02-25 by Aang23