SatDump/SatDump #779: Satdump crashes when SDR stopped

thawkes70, 2024-08-12:
When using an Airspy mini Satdump will crash when the SDR is stopped.  works fine when I use an RTL_SDR.  Tried deleting the settings.json, problems still persists, also tried the nightly builds and Windows portable versions.  Couldn't see anything untoward in the log file, I have attached it for reference

Hardware - Surface Pro Intel(R) Core(TM) i5-7300U CPU @ 2.60GHz  , 8Gb RAM

Version - Windows 1.2.0 and 1.2.1 nightly builds

[satdump-20240813T081819.log](https://github.com/user-attachments/files/16592529/satdump-20240813T081819.log)

JVital2013, 2024-08-13:
Hello!

To help narrow things down: do you have this issue when using other software, like SDR++?

thawkes70, 2024-08-14:
Hi,

Just tried it in SDR# and I could start and stop it with no problems.

t0nitogp, 2024-08-17:
Yep I just confirmed this, stopping Airspy Mini causes Satdump to crash on latest Nightly build, I had to reinstall an older version for it to work.

JVital2013, 2024-08-20:
This has been worked around in the latest Windows nightly. Hopefully a proper fix comes soon, after my Airspy mini gets here

JVital2013, 2024-08-22:
The permanent fix has been applied. Thanks for the report!

closed 2024-08-22 by JVital2013