SatDump/SatDump #780: satdump without ui need help

dl5raz, 2024-08-13:
I have problem to find out parameterstring : satdump ...?.. for Meteor wxsat
I use sdrplay on linumint, satdump-ui workes ok 
Want  using  satdump on a raspi4  without  ui to  reduce  workload  on  the raspi4

JVital2013, 2024-08-13:
Hello! Your command should be (roughly):

```
satdump live noaa_apt . --source sdrplay --samplerate 2e6 --frequency 137.1000e6 --satellite_number 19 --sdrpp_noise_reduction --source_id 190309B796
```
If that does not work, try it without the source_id parameter (in case there's something wrong with the ID; `satdump sdr_probe` will tell you what it should be.

Make sure to verify a valid samplerate in the UI as well. If I recall correctly, 1.024e6 is not one of the samplerates the sdrplay can handle (so I changed it to 2 in my example command)

You will also need to specify `--lna_gain` and `--if_gain`. SDRPlay SDRs don't have a general gain like it is often presented to the user for RTL-SDR, so you need to specify both of the individual gain controls, based on what works in the UI

dl5raz, 2024-08-13:
thanks a lot! have now a working command line:
/usr/bin/satdump live noaa_apt /home/mint/Bilder  --source sdrplay --samplerate 2e6 --frequency 137.100e6 --satellite_number 19 --sdrpp_noise_reduction  --lna_gain 9 --if_gain 54  --start_timestamp 1723576491 --finish_processing --timeout 978
next question: is there a way to use the internal satdump scheduler to make things automatic? 
example would be nice for the two meteor sats!

JVital2013, 2024-08-20:
Yes, autotrack can be used via the CLI. For now, you need to manually compose a .json file to tell it what to track, but soon a feature will be added to export CLI autotrack configurations from the UI.

You can read up on configuring the json manually at https://www.a-centauri.com/articoli/automating-satellite-reception-with-satdump-cli.

JVital2013, 2024-10-06:
The ability to export cli autotrack configs from the UI is now available in the latest nightlies, and the soon-to-be released SatDump 1.2.1

[image]

closed 2024-10-06 by JVital2013