SatDump/SatDump #782: Satdump crashes at random intervals - appears to be connected to SDR

PeteW841, 2024-08-15:
Satdump crashes/freezes at random intervals. Can be at any time - sat idle waiting for AOS, or halfway through decoding live NOAA pass. Makes no difference if audio is muted, or if SDR is idle

SDR - Genuine SDRplay RSP1A
Apple MacBook Pro M1

Satdump stable version 1.20

Relevant section of report below:

-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               satdump-ui [3422]
Path:                  /Users/USER/Desktop/SatDump.app/Contents/MacOS/satdump-ui
Identifier:            com.altillimity.satdump
Version:               1.2.0 (???)
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-08-15 11:11:11.5554 +0100
OS Version:            macOS 14.6.1 (23G93)
Report Version:        12
Anonymous UUID:        6A0FEE65-71A9-96E3-FB2B-7DA304F11746

Time Awake Since Boot: 12000 seconds

System Integrity Protection: enabled

Crashed Thread:        1

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000150
Exception Codes:       0x0000000000000001, 0x0000000000000150

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [3422]

VM Region Info: 0x150 is not in any region.  Bytes before following region: 4307680944
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      100c20000-100cc0000    [  640K] r-x/r-x SM=COW  /Users/USER/Desktop/SatDump.app/Contents/MacOS/satdump-ui

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1821fabac __ulock_wait + 8
1   libsystem_pthread.dylib       	       0x18223c48c _pthread_join + 608
2   libc++.1.dylib                	       0x182171aa8 std::__1::thread::join() + 36
3   satdump-ui                    	       0x100c2ea60 ctpl::thread_pool::stop(bool) + 244
4   satdump-ui                    	       0x100c49b18 ctpl::thread_pool::~thread_pool() + 32
5   libsystem_c.dylib             	       0x1820f72e8 __cxa_finalize_ranges + 476
6   libsystem_c.dylib             	       0x1820f7070 exit + 44
7   satdump-ui                    	       0x100c2dbe4 main + 5984
8   dyld                          	       0x181eaf154 start + 2476

Thread 1 Crashed:
0   satdump-ui                    	       0x100c8640c satdump::ViewerApplication::loadDatasetInViewer(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>) + 864
1   satdump-ui                    	       0x100c4c0e8 satdump::processing::process(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, nlohmann::json_abi_v3_11_2::basic_json<std::__1::map, std::__1::vector, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char>>, bool, long long, unsigned long long, double, std::__1::allocator, nlohmann::json_abi_v3_11_2::adl_serializer, std::__1::vector<unsigned char, std::__1::allocator<unsigned char>>, void>) + 2312
2   satdump-ui                    	       0x100c5dd60 std::__1::__packaged_task_func<satdump::RecorderApplication::stop_processing()::$_0, std::__1::allocator<satdump::RecorderApplication::stop_processing()::$_0>, void (int)>::operator()(int&&) + 596
3   satdump-ui                    	       0x100c41c48 std::__1::packaged_task<void (int)>::operator()(int) + 92
4   satdump-ui                    	       0x100c49798 ctpl::thread_pool::set_thread(int)::'lambda'()::operator()() const + 356
5   satdump-ui                    	       0x100c49508 void* std::__1::__thread_proxy[abi:ue170006]<std::__1::tuple<std::__1::unique_ptr<std::__1::__thread_struct, std::__1::default_delete<std::__1::__thread_struct>>, ctpl::thread_pool::set_thread(int)::'lambda'()>>(void*) + 52
6   libsystem_pthread.dylib       	       0x182239f94 _pthread_start + 136
7   libsystem_pthread.dylib       	       0x182234d34 thread_start + 8

PeteW841, 2024-08-15:
This morning I successfully received and decoded a NOAA pass. The satellite audio slowly dropped out, as expected, leaving the usual background hiss. After a few minutes, the frequency display stopped moving, and Satdump treats the SDR as being disconnected. A full restart of Satdump is then required

PeteW841, 2024-08-15:
Just to show how random the SDR disconnect is - whilst receiving a second NOAA pass, the Satdump & the SDR stopped talking to each other with around 3 minutes to go until LOS. When I clicked on "start" to try and restart the SDR, an error message appeared which is on the attached screenshot 
[image]

[image]

PeteW841, 2024-08-21:
Hi, thanks for taking the time to reply. No, it’s a short USB cable. I have an old Windows laptop with SatDump installed on it that runs for hours without any issues. As it is over 10 years old I was hoping to switch over to using my MacBook Pro

RAD750, 2024-09-30:
If you use a SDRPlay make absolutely sure that the cable is good (wiggle both ends, try also stretching or otherwise moving it with the SatDump running). From my experience, SDRplays absolutely hate even slightly defective cables and the API will lock up instantly.

JVital2013, 2024-09-30:
Along with what @RAD750 said, it is worth trying the latest nightly. It looks like the crashing thread is the one loading the dataset in the viewer, and there's a crash I recently fixed with that on macOS. 

Either way, let us know!

PeteW841, 2024-10-04:
RAD750 & JVital2013 - Thank you both for your replies. I installed the latest nightly build and have spent the last four days testing it. I'm pleased to say that there has been just one crash over many hours of testing, so I am also going to replace the USB cable with the best quality version I can find. It looks like the revised software version and the cable have both had a part to play.  Thanks again for your help, it is very much appreciated.

RAD750, 2024-10-04:
You're welcome! I personally use an Ugreen cable from amazon, it is cheap and works very well.

closed 2024-11-18 by JVital2013