SatDump/SatDump #785: tracking scheduler

brightkill, 2024-08-18:
Nice CLI update by Satdump, but I want to ask:
Is it possible to use tracking scheduler with CLI only (e.g. use it with Raspberry Pi via SSH)? If not, will it be implemented in future versions?

JVital2013, 2024-08-20:
Yes, autotrack can be used via the CLI. For now, you need to manually compose a .json file to tell it what to track, but soon a feature will be added to export CLI autotrack configurations from the UI.

You can read up on configuring the json manually at https://www.a-centauri.com/articoli/automating-satellite-reception-with-satdump-cli.

brightkill, 2024-08-20:
appreciate your post too much, big thanks

JVital2013, 2024-10-06:
The ability to export cli autotrack configs from the UI is now available in the latest nightlies, and the soon-to-be released SatDump 1.2.1

[image]

closed 2024-10-06 by JVital2013