SatDump/SatDump #795: Is it normal that SatDump crashes after processing a wav file ?

MoutonBinoclard, 2024-08-29:
I tried this tutorial to see the Noaa satellite picture  : https://www.a-centauri.com/articoli/noaa-poes-satellites-reception
however, when I click start, after processing the audio, the software just closes itself. Is it normal ?

zulufoxtrot, 2024-08-31:
I have the same issue.
Satdump 1.2.0 on macOS.
Satdump does process the file and does save it to the output directory.

zulufoxtrot, 2024-08-31:
Satdump also throws an error at startup: "Failed to load TLEs"
[image]
And the "tracking" window is blank:
[image]

JVital2013, 2024-09-09:
Hello everyone! No, crashing is not normal. We aren't the Windows Vista developers.

I'd recommend updating to the latest nightly (https://github.com/SatDump/SatDump/releases/tag/nightly) to see if that fixes it. If not, please share the log file from one of your crashes.

Thanks!

brandondowns93, 2024-10-19:
Hi all,

I was running into this same problem (on macOS 11.7.10, also processing a .WAV), and after installing the nightly build per @JVital2013's recommendation, I'm no longer getting a crash. Thanks!

I'm attaching a log of when 1.2.0 crashed last, if that's of any help?
[satdump-20241017T211853.log](https://github.com/user-attachments/files/17441863/satdump-20241017T211853.log)

zulufoxtrot, 2024-10-19:
Can confirm it no longer crashes on the nightly.

JVital2013, 2024-10-20:
Thanks for the reports guys - it sounds like this one is fixed so I'll close it for now. If it re-surfaces, feel free to open a new issue.

closed 2024-10-20 by JVital2013