SatDump/SatDump #797: Signal Lock

njrusty, 2024-09-01:
I am running vitality-goes and SatDump on my RP4. I used to use GOESTOOLS and with my now current system, I am not sure how to executed a linux command to see my data and how well I am locked onto GOES16. Getting some drops and may need to realign but not sure how to see the signal strength.

Appreciate any help.

alsldlflglhljlk, 2024-09-03:
You could use the SatDump GUI's live GOES HRIT decoder to aim your dish and find where the lowest BER is, and then use whatever other software you want to decode. Or you could SatDump itself to do all the decoding.

njrusty, 2024-09-04:
Could you please explain how this is done???

I log into my system via: http://192.168.1.8/Current-Weather and have looked their the menu and do not see anything such as GOES HRIT.

Appreciate any help!!!

alsldlflglhljlk, 2024-09-06:
Open the SatDump GUI
[image]

Go to the Recorder tab, select your SDR from the menu, set the frequency to 1694.1 MHz and click Start. I'm using an SDRPlay RSP1 here, but on RTL2832 based SDRs (RTL-SDR Blog), make sure the sample rate is 2.4 Msps and that the AGC checkbox is checked.
[image]

Scroll down to the Processing section and search for "GOES-R HRIT". Select it and click Start. When your antenna is aimed at a GOES satellite, it should look like this:
[image]

Pay attention to the BER (highlighted in the above screenshot). Try to minimize that number for the best signal quality.

alsldlflglhljlk, 2024-09-06:
Once you've minimized the BER, you can then close SatDump and use whatever other software you want for processing.
Or, if you leave SatDump in the above state, it will automatically begin processing and saving data and images.

njrusty, 2024-09-06:
Appreciate the effort. Unfortunately, my system is running on an RP4 and it appears these instructions above are for a PC based system.

alsldlflglhljlk, 2024-09-06:
The SatDump GUI should be available on the Raspberry Pi, but if not, you can install SatDump on a PC, connect your SDR to your PC, aim the dish for an ideal BER, and connect it back to your Raspberry Pi.

njrusty, 2024-09-07:
Appreciate the support, Linux still learning. I found the satdump-ui but when I launch it, it wants a program to launch it with. Not sure where to go from there

alsldlflglhljlk, 2024-09-08:
No, I meant run `satdump-ui` form the command line, but you could also do the same thing with goestools. You could use goestools to aim your dish and find the lowest error rate, and then use whatever other software you want after that.

mpentler, 2024-11-22:
@njrusty how are you getting on with this? I'm trying to help tidy up here.