SatDump/SatDump #801: lua compile issue

drforbin, 2024-09-06:
**Description of the issue**
I'm getting the following errors

[ 54%] Building CXX object plugins/noaa_metop_support/CMakeFiles/noaa_metop_support.dir/main.cpp.o
[ 54%] Linking CXX executable ../satdump
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: ../libsatdump_core.so: undefined reference to `lua_newuserdata'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: ../libsatdump_core.so: undefined reference to `lua_replace'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: ../libsatdump_core.so: undefined reference to `luaopen_bit32'
/usr/lib/gcc/x86_64-pc-linux-gnu/13/../../../../x86_64-pc-linux-gnu/bin/ld: ../libsatdump_core.so: undefined reference to `lua_insert'
collect2: error: ld returned 1 exit status
make[2]: *** [src-cli/CMakeFiles/satdump.dir/build.make:298: satdump] Error 1
make[1]: *** [CMakeFiles/Makefile2:1266: src-cli/CMakeFiles/satdump.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

JVital2013, 2024-09-08:
Hello!

Before we get too far, let's try deleting the contents of your `build` folder and start over again, starting with the initial cmake step. 

If that does not work, please let us know what you are using of the following:

- Operating system/OS Version
- Compiler/compiler version
- Cmake version
- Full cmake output

This is the first I have heard of this issue so I'm guessing it's either a corrupt build cache, or something specific to your system.

drforbin, 2024-09-09:
Hi. I got it to work linking to MY LUA 5.4. I have the diff in cmake, should I pos them here?
It seems the repo version is linking against a provided version of lua and the symbols refrenced above are absent ?

Let me know!

Thxs DF

JVital2013, 2024-09-09:
We're certainly open to improvements, but you're the first person to report these issues with Lua in SatDump. I'm very suspicious there's something else going on with your system, or you otherwise modified the code. The symbols you listed are included in the code as-is, and it works fine for the rest of our users as far as we know.

Feel free to share your patch - it's certainly welcome if there's an issue! But, we'll need to know all the items I previously asked for to validate if it's a general issue with a particular system configuration, or a problem specific to your system.

Thanks!

drforbin, 2024-09-11:
Ok,

Here is the diff for your review. 

```
diff --git a/src-core/CMakeLists.txt b/src-core/CMakeLists.txt
index 6004cf41..32ff4cfd 100644
--- a/src-core/CMakeLists.txt
+++ b/src-core/CMakeLists.txt
@@ -156,6 +156,19 @@ else()
     find_library(NNG_LIBRARY nng REQUIRED)
     target_link_libraries(satdump_core PUBLIC ${NNG_LIBRARY})

+
+    # LuaJIT
+    #    include(FindLuaJit)
+
+    #    if(LUAJIT_FOUND)
+    #    target_compile_definitions(satdump_core PUBLIC SOL_USE_LUAJIT="1")
+    #    target_link_libraries(satdump_core PUBLIC ${LUAJIT_LIBRARIES})
+    #    target_include_directories(satdump_core PUBLIC ${LUAJIT_INCLUDE_DIRS})
+    #    message("-- Building with LuaJIT!")
+    # else()
+    #    target_include_directories(satdump_core PUBLIC libs/lua)
+    #endif()
+
     if(BUILD_ZIQ)
         # zstd
         find_library(ZSTD_LIBRARY zstd REQUIRED)
@@ -163,8 +176,21 @@ else()
         target_compile_definitions(satdump_core PUBLIC BUILD_ZIQ="1")
     endif()

+
+    find_package(Lua REQUIRED)
+    target_link_libraries(satdump_core PUBLIC ${LUA_LIBRARIES})
+    target_include_directories(satdump_core PUBLIC ${LUA_INCLUDE_DIR})
+
+
+
+
     # Lua
-    target_include_directories(satdump_core PUBLIC libs/lua)
+    #    target_link_directories(satdump_core PUBLIC libs/lua)
+
```

FYI....The SATDUMP project is great! 
thank you for your effort

drforbin, 2024-09-12:
NOTE,

The important part is adding 

```
      find_package(Lua REQUIRED)

      target_link_libraries(satdump_core PUBLIC ${LUA_LIBRARIES})
      target_include_directories(satdump_core PUBLIC ${LUA_INCLUDE_DIR})
```

and commenting out (deleting)

```
target_link_directories(satdump_core PUBLIC libs/lua)
```

drforbin, 2024-09-16:
Should I post this is a pull request for Gentoo users?

JVital2013, 2024-09-25:
Hello, sorry for the delay.

I'm hesitant to push something like this, because it would require all systems to have lua installed, which isn't something we're requiring right now by using the bundled version. It can also be useful to ensure everyone is using the same version of Lua within SatDump.

We can go down this road if need-be, but I'd rather see if we can figure out why it doesn't build on Gentoo without this. I guess it's time for us to get our feet wet with Gentoo!

drforbin, 2024-09-27:
Sure.. Sorry I did not get back to you sooner! What would like me to do?

JVital2013, 2024-09-28:
I had some time and spun up a fresh Gentoo machine in an attempt to reproduce this issue, but unfortunately (?) it mostly worked fine for me:

[image]

There are some issues with installation that we need to fix up (it installs libs in /usr/lib, instead of /usr/lib64 when appropriate), but I'm not seeing the compile issue you're describing. Are you using Gentoo itself, or some other distribution based on it? Any special portage profile or other settings?

drforbin, 2024-09-28:
No, I'm using full gentoo linux. So what is it linking against?

JVital2013, 2024-10-06:
For Lua, SatDump's not linking against anything because Lua is included within SatDump - including all necessary symbols. This functionality has been verified several different ways from several people - including on Gentoo - so the issue seems to be with your system.

I've updated the README at https://github.com/SatDump/SatDump?tab=readme-ov-file#linux to include the required Gentoo dependencies for SatDump. From there, the normal Linux build instructions work. The cmake files have also been updated to more properly support systems like Gentoo.

I'd encourage you to review these and try building again, from scratch. If there are still errors, please give us the full output of the `cmake` command, the `make` command, and any other pertinent information (such as, which `eselect profile` you are using, and what make flags you have in /etc/portage/make.conf).

drforbin, 2024-10-06:
I will definitely pull the updates and try everything out and get back with you regarding the results.
Give me a few days.

JVital2013, 2024-10-06:
No worries, take your time!