SatDump/SatDump #802: Projection map alignment

67Robbo, 2024-09-07:
Is it possible to re-center the projection map here in New Zealand my images when projected split across the screen as attached.
[image]

Great software - Received Metop B & C today with ease.

alsldlflglhljlk, 2024-09-14:
Checking `Auto` and `Auto Scale` in the projection tab should show only your area.

RAD750, 2024-09-30:
As alsdlflglhljlk said, if you do this with the GUI you can select `Equirectangular` and then both `Auto` boxes, but if you do this automatically (e.g. on an automated station) you'll want to edit the 'satdump_cfg.json` according to [the documentation](https://docs.satdump.org/projections.html) 

If you're not familiar with how the composite system works, you can read [this documentation](https://docs.satdump.org/composites.html)

mpentler, 2024-11-22:
Trying to help get some of these closed. How are you getting on @67Robbo ?

closed 2025-10-23 by Cpt-Dingus