SatDump/SatDump #803: Possible decoder for GK-2A UHRIT

eswnl, 2024-09-07:
**What would you want to be added to satdump?**
Decoder for GK-2A UHRIT

**Describe the feature you'd like to have**
I can obtain sample transport streams for it, but it needs verifying/analysing.

**Describe alternatives you've considered**
I have the extractor for GK-2A UHRIT in SatDump, but no decoder

**Additional context**
[image]
Example header
[image]

Aang23, 2024-09-12:
Hi @eswnl! I've looked at your data, and it'll be simple enough. However I'll need a lot more continuous data if that is possible - say 30mins / 1h of raw TS to work off, as files take quite some time to go through.

eswnl, 2024-09-12:
Yes, I think it’s not continuous as packets are getting lost probably because of SNR. Much to my disappointment, I thought a demodulator lock was enough.

Do you happen to know how much dB is needed to get a solid reception if 6.6dB is the minimum level to get a lock?

Aang23, 2024-09-12:
No, I meant I just need a longer file than the few MB of TS you sent me last time to go further, it seemed perfectly fine otherwise! Sorry for the misunderstanding.

eswnl, 2024-09-13:
How many MB is required?

Aang23, 2024-09-13:
If you could record a single TS for ~30 mins or more that'd work.

eswnl, 2024-09-14:
I just recorded 2.5 hours worth. SNR ~ 7.0-7.5 dB and sent file to you. Comes to only 40 MB which I am confused as the link is 30 Mbps.

eswnl, 2024-11-02:
So I found out that the cadu frames are nested in the bbframes output in dvb-s2. I managed to isolate the cadu frames after some effort. I tried feeding the cadu into SatDump after adjusting the cadu length in the pipeline, but I got lrit crc errors. 
I think it’s because satdump is looking for the rs check bytes (128 bytes) appended to each cadu but UHRIT doesn’t use this and uses possibly 2 byte crc.

pluggerlockett, 2025-06-26:
Hi @Aang23.

Just thought I'd chime in here as well regarding the above decoder request. John has documented some further info on GK-2A UHRIT here.
[https://github.com/eswnl/GK-2A-UHRIT-extractor](url)

if you want some data to play with please let me know. I have both baseband and CADUs of varying sizes and my setup was averaging ~12dB SNR when I was recording.

[image]

sergsero, 2025-07-27:
Hi pluggerlockett,
> Hi [@Aang23](https://github.com/Aang23).
> 
> Just thought I'd chime in here as well regarding the above decoder request. John has documented some further info on GK-2A UHRIT here. [https://github.com/eswnl/GK-2A-UHRIT-extractor](url)
> 
Thank you for sharing the link to github.com.

In the text on that page in section 'Preparation': https://github.com/eswnl/GK-2A-UHRIT-extractor/tree/main?tab=readme-ov-file#preparation, the author mentions the use of scripts: "Copy the scripts: RemoveSpaceHeaders.py , new.py, Getsegment.py to this folder", but I don't see any scripts in the repository.

Have they been removed? Or has the author chosen not to make them publicly available?

Best regards,
sergsero