SatDump/SatDump #806: GeoTIFF without projections

JustZhenya, 2024-09-12:
**What would you want to be added to satdump?**
I would like to have ability to make geotiff without doing expensive projections.

**Describe the feature you'd like to have**
Projections are really expensive to make (noaa21 picture takes near hour to make on my CPU).
Also they lose quality after processing.
I've seen geotiffs that are rectangular, for example here http://leoworks.terrasigna.com/sample-data
Looks like regular satellite picture in picture viewer, but looks projected when opened in QGIS.
I don't know how does it work, but I believe that coordinates just being add to raw picture and projection is being made by QGIS. Correct me if I'm wrong and if it even possible.

**Describe alternatives you've considered**
I want satdump to make geotiffs like this http://leoworks.terrasigna.com/sample-data
Or this (although I can't open this pic in pic viewer so I'm not sure it's not projected) https://www.star.nesdis.noaa.gov/goes/fulldisk.php?sat=G16

Aang23, 2024-09-12:
The images you sent above are in fact reprojected UTM, not raw LandSat data.

[image]

As for the NOAA ones, there seems to be something wrong with them indeed, as I rarely ever managed to open them, but they by no means seem to be using GEOS. (Images you see from GOES are in fact re-projected to a standardized view meant to "look like it's from the POV of the satellite")

[image]

GEOS is not supported by GeoTIFF (see https://gis-lab.info/docs/geotiff-1.8.2.pdf) without using a very nasty hack `libproj` uses by default (hence disrespecting the very idea of GeoTIFF). Doing things this way would work with some software but I'm not a fan of doing hacky things like that with a standard.

Otherwise, the same goes for what you want with LEO images. It's simply not supported at all, and the projection process for LEO data being very complicated being the scene, you'd only be in for more pain.
It is possible to just add GCPs, but no software - even QGIS will work with that. It won't do the TPS transform, and if you use GDAL to do it externally, you just get worse and slower results than with SatDump's stuff.

One thing I can suggest to help is to try and get SatDump to do the projection on your GPU, as any decent-ish GPU will do the projection much, much faster. On my machine I'll do NOAA-21 in seconds even for a massive pass, and a full FY-3F DPT dump of an orbit in a few minutes.

JustZhenya, 2024-09-12:
Thank you for detailed answer.
So it is not possible and geotiff samples that I've downloaded are in fact already projected.
As for opencl, I've tried it and unfortunately AMD stopped support for opencl for my RX590. However I can try to use old drivers...

closed 2024-09-12 by JustZhenya