SatDump/SatDump #809: Processing crashes SatDump on nightly versions

ErdbeerbaerLP, 2024-09-18:
**Description of the issue**
Whenever the processing of a pass finishes, SatDump crashes. The processing itself seems to be unaffected by this, as the output files are generated.
Tested both with NOAA and METEOR attempts.
Happens both in offline processing, as well as in live processing.
When ran via `satdump-ui` in terminal, it additionally spits out `Segmentation fault (core dumped)`

**Hardware (SDR/PC/OS)**
NooElec NESDR SMArTee XTR via SDR++ Server hosted on a PC running TrueNAS (Based on Debian 12), no Desktop Env.
SatDump running on my Laptop; 32GB RAM; Intel i5-10300H; running Kubuntu 24.04.1 LTS

**Version (Eg, 1.0.0, CI Build 171)**
v1.2.1-1333698
Tested on multiple different nightly versions. Issue does **not** exist in stable release 1.2.0

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[satdump-20240918T081651.log](https://github.com/user-attachments/files/17042388/satdump-20240918T081651.log)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
can provide if needed

JVital2013, 2024-09-25:
Hello, thanks for reaching out.

Sorry this is happening to you. It looks like it is crashing when loading the dataset after decoding. Try going to Settings > User Interface and uncheck "Open Viewer after decoding", then save and restart SatDump.

If that works, try manually opening a dataset in the viewer. I'm guessing it will crash at this point, but let's just try this for now to see if we're on the right track.

ErdbeerbaerLP, 2024-09-26:
It indeed is the viewer. Disabling fixed the crash, and manually opening a dataset crashes.

JVital2013, 2024-09-26:
OK thanks - could you update to the latest nightly and try again? I just identified and fixed a crash that happens on some machines when opening datasets.

ErdbeerbaerLP, 2024-10-01:
Sorry for late reply. Just picked up the latest nightly, tried to manually import a few datasets and it did, in fact, not crash.

wx4cb, 2024-10-06:
i actually have the same issue with the windows nightly build from yesterday (1.2.1-1281ad5).

I will try disable the processing and see what happens.

wx4cb, 2024-10-06:
Update: yea turning that off allows it to carry on recording passes but i havent figured out the best way to "process" multiple files at the same time

JVital2013, 2024-10-06:
@wx4cb this sounds like a different crash from @ErdbeerbaerLP - whose issue sounds to be fixed. Can you please post a full SatDump log where it crashes so we can investigate?

JVital2013, 2024-10-14:
Since @ErdbeerbaerLP's issue seems to be resolved, I'm going to close this. If you're still having issues @wx4cb, open a new issue with a full log showing the crash. Thanks!

wx4cb, 2024-10-14:
Yea. Now the hurricane has gone by and i have power etc. back, im going to
look into it when i get home from work.

Leave it open for now if you dont mind and ill get you more info tonight

wx4cb, 2024-10-15:
[satdump-20241006T122127.log](https://github.com/user-attachments/files/17370561/satdump-20241006T122127.log)

here's a logfile from when it crashes... don't have the antennas up right now (gotta love hurricanes) but i was able to get at the log files on my computer. Not 100% what's going on, but maybe low signal strength as i did have the antennas down low due to the coming hurricane. once i get the antennas back up i will check again. but any advice would be appreciated.

JVital2013, 2024-10-20:
First off: glad to hear you're safe from the hurricane and all! I'll hold a moment of silence for those antennas, though.

Odd - this log is showing that it crashed before it would have hit auto processing. It crashed when it was still saving the raw data, so either that caused it, or something else at the same time. For now, try turning auto processing back on.

Do you have "Stop SDR when IDLE" enabled in the scheduler? That's the next thing I would try turning off for testing, if you have it enabled

mpentler, 2024-11-22:
In an attempt to help some of these issues get closed - ping @wx4cb :)

closed 2026-09-30 by Aang23