SatDump/SatDump #813: Composite System

henkthoma, 2024-09-20:
I would like some composites to be generated and some not, so configured ~/.configure/satdump/settings.json according to the 'Overriding the default configuration' section at the bottom of https://docs.satdump.org/composites.html. However, still the same composites are automatically generated, and any instructions "autogen": true (or false) are ignored. What could be the problem?

I have included my settings.json (with changed latitude, longitude and elevation for privacy).

[settings.json](https://github.com/user-attachments/files/17071118/settings.json)

JVital2013, 2024-09-25:
Hello @henkthoma!

Thanks for reaching out. I noticed two things that may be causing your issue:

1. The path for the file is usually `~/.configure/satdump/settings.json`, not `configure`. The SatDump log will say for sure, near the top
2. You're missing the `rgb_composites` level of the settings config. It should look like this (omitting all the other stuff for brevity):

```
    "viewer": {
        "instruments": {
            "avhrr_3": {
                "rgb_composites": {
                    "MCIR": {
                        "autogen": false
                    },
                    "MSA (Uncalibrated)": {
                        "autogen": false
                    }
                }
            }
        }
    },
```

I hope this helps!

henkthoma, 2024-09-27:
Yes my path is: ~/.config/satdump/settings.json, not configure, I mentioned it wrong.
I tried adding rgb_composites and got similar results. After that I even wrote it like:
"avhrr_3": {
                "handler": "image_handler",
                "name": "AVHRR/3",
                "rgb_composites": {

But still similar results. Updating the satdump_cfg.json (instead of settings.json) itself also didn't make any change.

JVital2013, 2024-09-30:
Indeed, that is a mistake in our docs. Thanks! I've updated it on the repo that controls the website, and the docs website itself should be updated soon.

closed 2024-09-30 by JVital2013