SatDump/SatDump #816: Capture only images

njrusty, 2024-09-22:
Is there a way or script to only capture images and ignore all text downloads???

JVital2013, 2024-09-23:
Are you referring to GOES-R HRIT?

If so, take a look in the pipelines/GOES.json file. In here, you can find the following lines under goes_hrit:

https://github.com/SatDump/SatDump/blob/5c81025d60bcb1622d9aa58a927b1cc3b13dcc82/pipelines/GOES.json#L154-L159

Simply turn all of them off other than write_images, and the only thing saved will be the images

njrusty, 2024-10-11:
Modified the script but still capturing all text files. Oh well, was a try.

JVital2013, 2024-10-14:
I have recently verified that setting `write_emwin: false,` does turn off saving text files on my end. Can you please share your full SatDump log after changing this setting, and attempting a GOES decode?

njrusty, 2024-10-14:
[GOES.json](https://github.com/user-attachments/files/17369635/GOES.json)

Not sure where that log file is located but I have my GOES.json attached

mpentler, 2024-11-22:
> [GOES.json](https://github.com/user-attachments/files/17369635/GOES.json)
> 
> Not sure where that log file is located but I have my GOES.json attached

Look in your SatDump config directory

njrusty, 2024-11-23:
Attached is a sceenshot of my SatDump folder. There is no config directory.
[image]

mpentler, 2024-11-23:
@JVital2013 a little help pls - where does the log go on Linux. I assume this is a screenshot of their build folder.

mpentler, 2024-11-23:
> Attached is a sceenshot of my SatDump folder. There is no config directory. ![satdump](https://private-user-images.githubusercontent.com/150564341/389213819-b9a964bb-04ea-4c69-a4d2-a671a75e146f.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIzODYyMTksIm5iZiI6MTczMjM4NTkxOSwicGF0aCI6Ii8xNTA1NjQzNDEvMzg5MjEzODE5LWI5YTk2NGJiLTA0ZWEtNGM2OS1hNGQyLWE2NzFhNzVlMTQ2Zi5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyM1QxODE4MzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jYzYwMmVmMDdjZTk1YzE0MmYxYjIyNzdkNTBjNjE4YzZjOTlkYzlhNzMxYzc0MDY2ODZjMzBlNDYwMTkxNzU1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.0HIM8M9RdpxXAzWHqJc7nLLuIg7DOYQun1uDDTUtpAc)

Apologies. Biased towards my own experience. This should help.

https://docs.satdump.org/faq_troubleshooting.html#where-are-satdump-logs