SatDump/SatDump #817: SatDump (nightly) stops responding on launch

jcv09, 2024-09-22:
**Description of the issue**
When launching SatDump, the app stops responding after '0 TLEs loaded' message.

**Hardware (SDR/PC/OS)**
2023 Dell XPS 15, i7 rtx 4070, windows 11

**Version (Eg, 1.0.0, CI Build 171)**
Nightly

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[satdump-20240922T115331.log](https://github.com/user-attachments/files/17091067/satdump-20240922T115331.log)

JVital2013, 2024-09-23:
Try removing the DLL located at C:\Program Files\SatDump\lib\satdump\plugins\remote_sdr_support.dll, and see if that fixes the issue for you

jcv09, 2024-09-23:
There don't seem to be any files with that name there, could they have installed somewhere else?

JVital2013, 2024-09-23:
According to the log file you uploaded, the file is there:

[image]

jcv09, 2024-09-24:
My bad, I was looking for it in C:\Program Files\SatDump\plugins instead of C:\Program Files\SatDump\lib\plugins. It launches now, thank you!

closed 2024-09-24 by jcv09