SatDump/SatDump #819: make error

david-196610, 2024-09-23:
During make process I got this error

SatDump/src-core/common/utils.cpp:89:53: error: ‘CURLSSLOPT_NATIVE_CA’ was not declared in this scope
   89 |         curl_easy_setopt(curl, CURLOPT_SSL_OPTIONS, CURLSSLOPT_NATIVE_CA);

SatDump/src-core/common/utils.cpp:126:53: error: ‘CURLSSLOPT_NATIVE_CA’ was not declared in this scope
  126 |         curl_easy_setopt(curl, CURLOPT_SSL_OPTIONS, CURLSSLOPT_NATIVE_CA);

And make stop.
I'm using UBUNTU 20.04 
I follow the compiling istructions installing all dependance.I particolar CURL is properly installad
Thank you for Help!

David

JVital2013, 2024-09-25:
Hi David,

This should be fixed as of the latest commit. Let us know if you still have issues!

The problem is caused by an old version of libcurl in Ubuntu 20.04, which does not have a feature we use. However, the feature is not needed on Linux (only Android at the moment), so we can safely disable those lines if the feature is not available.

Thanks for the report!

closed 2024-09-25 by JVital2013

david-196610, 2024-09-27:
Hi Jamie

Thank you.
Problem solved!
It work good!

Best regards

David