SatDump/SatDump #824: Could Not Decode GK-2A HRIT Demo ("LRIT CRC is invalid... skipping...")

FCWYzzr, 2024-09-27:
# What
Whatever method I use, it simply could not decode any image of it
# How
|name|value|
|---|---|
|pipeline|GK-2A HRIT|
|input level| tried all of them(backbase, soft, cadu)|
|pipeline argument|default(cs8, 6Msps, blank, blank, 0Hz)|
|result|"LR CRC is invalid... skipping..." x 2 & a empty gk2a_hrit.cadu file|
# Env
Window10 x86
SatDump 1.2.0, both offical portable & my build(except my build refuse to give any error log)
[COMS_Decryption_Sample_Cpp](http://nmsc.kma.go.kr/resources/enhome/resources/satellites/coms/COMS_Decryption_Sample_Cpp.zip)/IMG_ENH_37_IR1_20110414_110920_01.hrit
# detail
[logs](https://pastebin.com/L3waxNnr)
# probable reason
its my first time stepping in satellite sdr & I dont know how to decode it :(

eswnl, 2024-09-27:
I suppose I can help here. Have you extracted the decryption key and put it in the correct folder in SatDump?
If this is not done correctly, you will get a blank image.
Also what helped me was lowering the pll_bw in the pipeline to 0.002 to get solid deframer sync.
You need an SNR of at least 4.0 to get an image generally.

Also you can do it live, no need to record. Try changing the gain as well.

FCWYzzr, 2024-09-27:
> I suppose I can help here. Have you extracted the decryption key and put it in the correct folder in SatDump? If this is not done correctly, you will get a blank image. Also what helped me was lowering the pll_bw in the pipeline to 0.002 to get solid deframer sync. You need an SNR of at least 4.0 to get an image generally.
> 
> Also you can do it live, no need to record. Try changing the gain as well.
> 
> Also “ENH” looks like it was from the former COMS-1 satellite, might not decode here.

well, im using offline mode, trying to docode that demo file. i'm sure i've done with the key part (resource/gk2a/EncryptionKeyMessage.bin) as a former issue says. but i dont even get a blank image, it just logs error and exit, leaving a empty file & a empty folder.

eswnl, 2024-09-27:
I don’t think SatDump can process the hrit file, it’s probably at the wrong layer.

JVital2013, 2024-09-29:
Hello!

@eswnl is correct: SatDump cannot decode singular lrit/hrit files at this time. It can decode from baseband, soft symbols, or a .cadu packet stream

FCWYzzr, 2024-09-29:
Okay I'll accept that as the answer

closed 2024-09-29 by FCWYzzr