SatDump/SatDump #826: Satdump on Raspberry Pi doesn't start on the correct frequency

redaters, 2024-09-29:
When starting Satdump in CLI or GUI mode on the Raspberry Pi, the starting frequency isn't correct. In the GUI, when you change the frequency the SDR sets to the correct frequency. The CLI version can't change the frequency while running. This problem occurs every time that Satdump is run.

Raspberry Pi 4B, RTL-SDR Blog v3, externally powered USB hub.

Satdump 1.2

[satdump-20240929T174953.log.gz](https://github.com/user-attachments/files/17181493/satdump-20240929T174953.log.gz)

https://github.com/user-attachments/assets/8bd263b7-d440-4a47-8e4b-7576e08812cf

JVital2013, 2024-09-29:
Hello!

Thanks for the report. Could you please try the latest nightly version and let us know if you still have the same bug? https://github.com/SatDump/SatDump/releases/tag/nightly

redaters, 2024-09-29:
> Hello!
> 
> Thanks for the report. Could you please try the latest nightly version and let us know if you still have the same bug? https://github.com/SatDump/SatDump/releases/tag/nightly

@JVital2013 I tried this version but I'm having the same issue.

JVital2013, 2024-10-14:
Thanks for the reply (and sorry for my late one!) Can you try it without a USB hub? SDRs in general do not like to work correctly over a USB hub - or so it is frequently reported.

redaters, 2024-10-15:
I have the problem both with and without a USB hub. I originally connected my SDR without a USB hub but when I noticed the problem I connected the hub hoping to give it more external power.

JVital2013, 2024-11-18:
Hello,

Can you please update to the latest nightly and try again? Since the last time we chatted I made the RTL-SDR source more robust.

If it still does not work, please include a new log so we can investigate. More info should be logged now if there are issues.

nathan-42, 2025-06-02:
I came to report this exact same bug. Glad I checked before making my own video and posting. Will follow up shortly with a log.

Cpt-Dingus, 2025-10-23:
This is caused by a bug in librtlsdr, fixed in #1051

closed 2025-10-23 by Cpt-Dingus