SatDump/SatDump #828: Crash Saving Settings on Nightly Build

Arizona-Steve, 2024-09-30:
**Description of the issue**
Clicking on the save button in the settings tab causes the cursor to spin for about 2 seconds, then SatDump closes. The settings changes are not present after reopening SatDump. The Official 1.2.0 version does save settings properly, but it won't load the TLEs. I can manually adjust the settings values by editing "C:\Program Files\SatDump\share\satdump\satdump_cfg.json", but that doesn't seem like a good workaround.

**Hardware (SDR/PC/OS)**
RSP1
occurs on both a LivaZ2 and an I9 homebuilt
both running Windows11

**Version (Eg, 1.0.0, CI Build 171)**
SatDump v1.2.1-be975cf

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)
[satdump-20240930T104054.log](https://github.com/user-attachments/files/17194126/satdump-20240930T104054.log)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**
Available on request

JVital2013, 2024-09-30:
Hello!

Thanks for reaching out. To rule out anything simple, revert satdump_cfg.json to the original file, then delete %appdata%\satdump\settings.json.

Afterwards, try re-opening SatDump and clicking save. If it works, try editing your settings and save again. If the save crashes at this point, let us know what settings you changed.

Arizona-Steve, 2024-09-30:
I have not modified satdump_cfg.json since I installed today"s nightly build,
I deleted "C:\Users\Steve\AppData\Roaming\satdump\settings.json"
{save] without changing anything closed SatDump but a settings.json file was created
I opened SatDump and changed GeneralSatDump/Default QTH Label and clicked [Save}
SatDump closed and the setting change was not saved.

Thank you for your support
Steve

Arizona-Steve, 2024-10-01:
Saving settings is now working in the current nightly build after trying the following workaround:
I have only tested by changing the QTH Name so far.

I installed the official 1.2.0 version on a test machine and modified the settings as described int the following tutorial:
https://usradioguy.com/satdump-for-meteor-noaa-decoding/
I transferred the resulting settings.json file to my weather machine, running today's nightly build (v1.2.1-be975cf).
Saving settings now worked on the nightly build and the changed value was retained across restarts of SatDump.

I changed the following settings in the official 1.2.0:
 &nbsp; &nbsp; &nbsp;  General SatDump
 &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;Latitude
 &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;Longitude
 &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;Altitude
 &nbsp; &nbsp; &nbsp;User Interface
 &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;Unchecked Open viewer after decoding
 &nbsp; &nbsp; &nbsp;Output Directories
 &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;Changed them all
 &nbsp; &nbsp; &nbsp; &nbsp; &nbsp; &nbsp;NOTE: Live Processing Directory was red and had both forward and backward slashes in the path.

Here is my modified settings.json:
[settings.json](https://github.com/user-attachments/files/17200058/settings.json)

Thank you for your support
Steve

JVital2013, 2024-10-01:
OK, it looks like we can reproduce the issue. We'll get back to you when it is fixed.

Thanks again for the report!

JVital2013, 2024-10-01:
This issue should now be fixed. It will take about 20 minutes from this writing for the nightly build to update with the fix.