SatDump/SatDump #829: Error on Nightly Build

waleed1668, 2024-10-01:
**In offline processing get this message (Unknown EUMETSAT file type! FCI-1C-RRAD-HRFI-FD)**

**PC**

**Version (1.2.1)**
[image]

JVital2013, 2024-10-14:
Hello @waleed1668!

Support for HRFI (High Resolution) archives has been fixed in the latest nightly. It will be ready in about 20 minutes. Let us know if you have any other issues!

closed 2024-10-14 by JVital2013

waleed1668, 2024-10-15:
> Hello @waleed1668!
> 
> 
> 
> Support for HRFI (High Resolution) archives has been fixed in the latest nightly. It will be ready in about 20 minutes. Let us know if you have any other issues!

**Yes, it's working**
Thank you

commit aa6e54a4 (2024-10-14): Fix #829