SatDump/SatDump #831: Debian build not working except on fresh install of Bookworm - missing dependancies / they won't load

Hawkeye6323, 2024-10-07:
**Description of the issue**  Debian installation problem - missing dependancies. Follow directions for installation of dependancies and several are not found (assumed out of date). 

**Hardware (SDR/PC/OS)**  Raspberry pi 4. Fresh Bookworm (version 12) install.

**Version (Eg, 1.0.0, CI Build 171)** 64-BIT ARM version from current link.  DEB file

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)  Unsure how to pull this up.

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**  Please see Jeremy Pierce recent posts on Ham Radio Crash Course Facebook page for screen shots. I can email them with an address.

JVital2013, 2024-10-08:
Hello,

Similar to your other post, please try the latest nightly. Our Raspberry Pi builds are built with Raspberry Pi OS Bookworm, 64-bit version.

I'd recommend uninstalling the current version, download the nightly deb, and run with `sudo apt install ./satdump_rpi64_latest_arm64.deb`.

If there are errors on the install, please post them here. Thanks!

Hawkeye6323, 2024-10-08:
Ok, I'll give that a try tonight.  Thank you for the timely response.  This looks like an awesome piece of software.  I follow "Save it for Parts" on YouTube and he uses this program a lot.  Thank you for all you do in support of it. 

Jeremy PierceKM6QAQ

Hawkeye6323, 2024-10-09:
I'm getting this message at the end of the install. I deleted the old version, downloaded the new and tried the sudo apt install command with the path to the Deb file in my downloads file folder. I don't believe it likes this. Probably need to extract it to the ./ folder I assume?  Thanks
Jeremy

Sent from AOL on Android

JVital2013, 2024-10-09:
Sorry, I'm not seeing any logs or anything. If you had it as an attachment it probably got lost. Can you post the log directly?

Hawkeye6323, 2024-10-09:
Hey Jamie, I haven't accessed the logs as you mentioned yet.  Last night I downloaded the nightly update to my cleanly installed Bookworm sd card.  Installed the package by right clicking on the downloaded file and everything installed no problem. 

I went back to the build-a-pi (hat tip KM4ACK's ham radio suite) sd card and tried doing the same.  I couldn't uninstall it as it wouldn't come up in software preferences.  Upon verification that I wanted to install the package up came a caution message that it couldn't find the package components - very weird.  I had previously tried to install the old file version via the command prompt with no success. There was a file folder created for SATDUMP which I attempted to delete and this had no effect.  Multiple attempts at this install with the same result last night. I'll try to pull logs tonight.  It's been a hot minute since I tried to SSH into the pi via putty so I can't guarantee success / might take me a few nights to get back in the saddle if you will. 

Thank you for your assistance in the matter.  
 Jeremy

Hawkeye6323, 2024-10-10:
> Hi Jeremy,
>
> Thanks for the reply. I'm glad to hear your bookworm SD card is working correctly. For your build-a-pi SD card, which version of Debian is that running? You can find that by running lsb-release -a.
>
> Our arm64 deb are currently only for Bookworm. If it's older, you will need to build SatDump yourself instead of installing the deb. You can follow the build instructions at https://github.com/SatDump/SatDump. Install the dependencies using the commands in the "Install Dependencies - Debian, Ubuntu, and other Debian-based distros" section, then the "Build SatDump" instructions under that.
>
> And no worries about the time it takes to get back into the saddle - we've all been there! 😊

Same version, Debiain Bookworm version 12.  

I tried building it last night after the auto install couldn't find the packages.  

I check that link you shared but I'm pretty sure it is what I used.  The a good deal of the dependancies would not install in C: and reported unfound.  

I probably screwed up after noting that.  I deleted CMAKE in software preference uninstall.  Then spent the rest of the evening trying to reload it.  

The CMAKE lists ARM Linux versions.  I tried the x86 version and am not able to get the ./bootstrap function to work. Perhaps wrong Linux file.  

Well hey I haven't given up, and I'm learning in the process. 

Jeremy