SatDump/SatDump #832: Debian (Raspberry Pi 4) TLE's will not initial install.

Hawkeye6323, 2024-10-07:
**Description of the issue**  Debian installation problem - TLE's won't initially install. Assume NORAD website changed the link.  

**Hardware (SDR/PC/OS)**  Raspberry pi 4. Fresh Bookworm (version 12) install.

**Version (Eg, 1.0.0, CI Build 171)** 64-BIT ARM version from current link.  DEB file

**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)  Unsure how to pull this up.

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**  Please see Jeremy Pierce recent posts on Ham Radio Crash Course Facebook page for screen shots. I can email them with an address.

JVital2013, 2024-10-08:
This sounds like the bug in 1.2.0, that has since been fixed in the latest nighties (https://github.com/SatDump/SatDump/releases/tag/nightly). Please update to that version and let us know. If you still have the issue on the nightlies, you can get the log by:

- SSH into the Pi
- `cd ~/.config/satdump`
- Type `ls` to list all files. Find the one with the newest date
- Type `cat <logname>` to display the log, which can be copied and pasted

Hawkeye6323, 2024-10-08:
Thank you for your assistance and timely response.  I'll give this a try and report my finding ASAP. 

Jeremy PierceKM6QAQ

JVital2013, 2024-11-17:
Hi @Hawkeye6323,

Are you still having this issue?

Hawkeye6323, 2024-11-21:
Ya, I walked away from it.  Might try again this weekend. 

JeremyKM6QAQ73