SatDump/SatDump #833: Latest nightly build does not find RTL-SDR or Airspy USB devices

TechMindsYT, 2024-10-07:
**Description of the issue**
Latest nightly build does not find RTL-SDR or Airspy USB devices.
**Hardware (SDR/PC/OS)**
PC
**Version (Eg, 1.0.0, CI Build 171)**
Nightly Build
**Logs after the crash** (satdump.log from ~/.config/satdump or %appdata%\satdump)

**Other info (Eg, Screenshots) / Files useful for debugging (CADU, etc)**

JVital2013, 2024-10-08:
Hello!

Which operating system are you on? Also, can you provide a full log of SatDump opening up?

TechMindsYT, 2024-10-08:
Windows 11

Here is the log:

[07:55:45 - 08/10/2024] (E) Usage : C:\Users\freak\Desktop\SatDump-Windows_x64_Portable\satdump-ui.exe [downlink] [input_level] [input_file] [output_file_or_directory] [additional options as required]
[07:55:45 - 08/10/2024] (E) Extra options (examples. Any parameter used in modules can be used here) :
[07:55:45 - 08/10/2024] (E)  --samplerate [baseband_samplerate] --baseband_format [cf32/cs16/cs8/wav16] --dc_block --iq_swap
[07:55:45 - 08/10/2024] (D) Starting with OpenGL 4.6.0 NVIDIA 565.90
[07:55:45 - 08/10/2024] (D) Max texture size: 32768
[07:55:45 - 08/10/2024] (I)    _____       __  ____                      
[07:55:45 - 08/10/2024] (I)   / ___/____ _/ /_/ __ \__  ______ ___  ____ 
[07:55:45 - 08/10/2024] (I)   \__ \/ __ `/ __/ / / / / / / __ `__ \/ __ \
[07:55:45 - 08/10/2024] (I)  ___/ / /_/ / /_/ /_/ / /_/ / / / / / / /_/ /
[07:55:45 - 08/10/2024] (I) /____/\__,_/\__/_____/\__,_/_/ /_/ /_/ .___/ 
[07:55:45 - 08/10/2024] (I)                                     /_/      
[07:55:45 - 08/10/2024] (I) Starting SatDump v1.2.1-b99cd26
[07:55:45 - 08/10/2024] (I) 
[07:55:45 - 08/10/2024] (I) Loading config satdump_cfg.json
[07:55:45 - 08/10/2024] (I) Loading user settings ./config/settings.json
[07:55:45 - 08/10/2024] (I) Loading plugins from ./plugins
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\aaronia_sdr_support.dll...
[07:55:45 - 08/10/2024] (E) Error loading ./plugins\aaronia_sdr_support.dll! Error : ".\plugins\aaronia_sdr_support.dll": The specified module could not be found. => src-core\core\plugin.cpp:19
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\aim_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin aim_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\airspyhf_sdr_support.dll...
[07:55:45 - 08/10/2024] (E) Error loading ./plugins\airspyhf_sdr_support.dll! Error : ".\plugins\airspyhf_sdr_support.dll": The specified procedure could not be found. => src-core\core\plugin.cpp:19
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\airspy_sdr_support.dll...
[07:55:45 - 08/10/2024] (E) Error loading ./plugins\airspy_sdr_support.dll! Error : ".\plugins\airspy_sdr_support.dll": The specified procedure could not be found. => src-core\core\plugin.cpp:19
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\analog_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin analog_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\aws_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin aws_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\bladerf_sdr_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin bladerf_sdr_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\bluewalker3_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin bluewalker3_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\cluster_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin cluster_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\cpp_compos_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin standard_cpp_compos loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\cubesat_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin cubesat_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\dmsp_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin dmsp_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\dscovr_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin dscovr_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\dvb_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin dvb_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\elektro_arktika_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin elektro_arktika_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\eos_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin eos_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\fengyun2_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin fengyun2_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\fengyun3_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin fengyun3_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\fengyun4_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin fenyun4_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\gcom_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin gcom_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\geonetcast_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin geonetcast_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\gk2a_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin gk2a_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\goes_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin goes_support loaded!
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\hackrf_sdr_support.dll...
[07:55:45 - 08/10/2024] (E) Error loading ./plugins\hackrf_sdr_support.dll! Error : ".\plugins\hackrf_sdr_support.dll": The specified procedure could not be found. => src-core\core\plugin.cpp:19
[07:55:45 - 08/10/2024] (T) Loading plugin ./plugins\himawari_support.dll...
[07:55:45 - 08/10/2024] (T) Plugin himawari_support loaded!
[... 534 lines cut ...]
[07:55:46 - 08/10/2024] (D)  - tianwen1_link
[07:55:46 - 08/10/2024] (D)  - timed_dump
[07:55:46 - 08/10/2024] (D)  - tubin_x_dump
[07:55:46 - 08/10/2024] (D)  - umka_1_dump
[07:55:46 - 08/10/2024] (D)  - veronika_link
[... cut ...]
[07:55:49 - 08/10/2024] (D) Device File Source
[07:55:49 - 08/10/2024] (D) Device Network Source
[07:55:49 - 08/10/2024] (D) Device PlutoSDR IP
[07:55:49 - 08/10/2024] (D) Device RTL-TCP
[07:55:49 - 08/10/2024] (D) Device SDR++ Server
[07:55:49 - 08/10/2024] (D) Device SpyServer
[07:55:49 - 08/10/2024] (T) FFT Rate 8333, Samplerate 1000000, Final Size 8192, Skip 141
[07:55:56 - 08/10/2024] (I) Saving user config at ./config/settings.json
[07:55:56 - 08/10/2024] (I) UI Exit
[07:55:56 - 08/10/2024] (I) Exiting!

JVital2013, 2024-10-08:
Thank you for the log. It looks like you're using the portable version, and the DLLs are messed up - they are probably mismatched versions. Perhaps you extracted the new version into a folder that already had an old version in it?

Please try doing a fresh extraction of the portable zip into a new folder and run that. Alternatively, try using the installer.

JVital2013, 2024-10-08:
Reviewing the logs again, I have another idea - if updating the vc redist does not work, we should verify it is loading libusb from the correct area. It really should be, but all the DLLs showing errors in your log are relying on a newly-patched version of libusb, so maybe it's pulling libusb from another directory.

The easiest way to do that is to download procmon - https://learn.microsoft.com/en-us/sysinternals/downloads/procmon. In the program, filter it to only show "Load Image" operations, by the process name "satdump-ui.exe":

[image]

Then, launch satdump and check libusb-1.0.dll's path. If it is anything other than SatDump directory, that is likely the problem.

[image]

TechMindsYT, 2024-10-08:
[image]

TechMindsYT, 2024-10-08:
I just coped the libusb from the portable nightly to c:\windows\system32 and now it works. Just hope it hasnt broken any other apps but will check.

JVital2013, 2024-10-08:
Interesting - there should not be a libusb-1.0.dll in System32. I'm guessing some other program put it there, but it's clearly overriding other libusb DLLs on your system, which is likely bad for more than just SatDump (even if it hasn't come up yet)

Removing it is probably the best call, then work out where it came from.

TechMindsYT, 2024-10-08:
I have endless amounts of SDR and Ham Radio related software on my computer, figuring where it come from will be a mission.

Maybe ZaDig installed it in System32, who knows..

JVital2013, 2024-10-08:
It's not Zadig - it doesn't do that. So that reduces it by one 😊.

I just did a test and confirmed that libusb-1.0.dll in system32 does indeed override libusb for other programs on the computer, which is very bad. I can see this behavior happening in other programs as well. It doesn't really look like an issue with SatDump: this is Windows behavior. It's only affecting SatDump for you because we aren't using the standard libusb dll anymore, but other programs can accept the system DLL. We likely haven't heard about it because some obscure program installed it on your system, that most people don't have.

I'll see if there's something I can do to prevent it from loading the DLL from system32. I'm not sure there is but it is worth a look.

JVital2013, 2024-10-08:
This has been fixed, and the fix will be available in the latest nightlies in about 20 minutes. It turns out I was wrong: this is not default Windows behavior, but behavior explicitly defined in dlfcn-win32, which we use to load the plugins. I patched dlfcn-win32 to prefer libraries in the local directory.

Thanks very much for reporting this, and for working with us to find a fix. This is an odd one that could have gone a long time without being solved, so I'm grateful for your help!

closed 2024-10-08 by JVital2013

TechMindsYT, 2024-10-08:
You are most welcome and glad to be of help. I have not yet featured SatDump on my YT channel as of yet, well, not for decoding images, but hopefully this weekend I will have a video out showcasing it being used to decode HRPT. Thanks again for your hard work, it's a great application.

commit 49bf7475 (2024-10-08): Fix #833